openwrt/luci #6468: luci-app-statistics: wireless stats only show quality stats after auto-refresh

imkuang, 2023-07-15:
## Steps to reproduce:

1. go to: Statistics → Setup → Network plugins tab, enable Wireless plugins and config to monitor a wireless interface.
2. go to: Statistics → Graphs -> Wireless tab, you should see four stats charts of quality, dbm, rate, and stations.
4. select refresh every 5 seconds and apply interval.
5. try toggling the stats tab and you'll see the wireless tab change from 4 charts to one after auto-refresh.

## Actual behavior:

Wireless stats charts display when not automatically refreshed:
[image]

display after auto-refresh with only quality stats one chart:
[image]

## Expected behavior:

The page should still show the four stats charts after auto-refresh

## Additional Information:

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03-SNAPSHOT'
DISTRIB_REVISION='r20183-80a6b0a917'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03-SNAPSHOT r20183-80a6b0a917'
DISTRIB_TAINTS='no-all'
```

BTW, there is no problem with tabs such as processor that also show multiple stats charts

jow-, 2023-07-26:
Can't reproduce here, probably specific to single radio devices.

imkuang, 2023-07-26:
Hi,

 

I'm actually running OpenWrt in a pve VM with a passthrough PCI device of Intel AX210 NIC.

 

Should it just be a front-end issue that all stats charts display correctly when not auto-refreshed?

I'm not familiar with luci at all, maybe when I have enough time I can go and learn about it.

jow-, 2023-07-26:
It is a frontend issue, yeah. It actually is supposed to only show a single chart, along with a secondary tab row to show radio specific detail charts:

[image]

Seems in your case the logic is confused, need to find some time to replicate it in a test setup.

PR #7344 (2024-10-24): luci-app-statistics: render all graphs on refresh if no summary
In the past, a summary page would usually be generated for the data, but this did not occur when there was only a single instance (e.g., a single Wi-Fi interface). As a result, the refresh mistakenly requested the summary instead of the available data.

i.e. graphs would mysteriously disappear on refresh.

See #6468 for more context.

- [x] This PR is not from my *main* or *master* branch :poop:, but a *separate* branch :white_check_mark:
- [x] Each commit has a valid :black_nib: `Signed-off-by: <my@email.address>` row (via `git c [... cut]