openwrt/luci #6483: luci-app-openvpn: not working show error last openwrt 23.05.0-rc2

henok400, 2023-07-26:
I have been trying to install openvpn at openwrt v23.05.0-rc2 that got error 
```
Installing luci-app-openvpn (git-23.013.73089-25983b9) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0-rc2/packages/arm_cortex-a9_vfpv3-d16/luci/luci-app-openvpn_git-23.013.73089-25983b9_all.ipk
Installing libnl-tiny2023-07-01 (2023-07-01-d433990c-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0-rc2/packages/arm_cortex-a9_vfpv3-d16/base/libnl-tiny2023-07-01_2023-07-01-d433990c-1_arm_cortex-a9_vfpv3-d16.ipk
```
the output error is 
```
Collected errors:
 * check_data_file_clashes: Package libnl-tiny2023-07-01 wants to install file /usr/lib/libnl-tiny.so
	But that file is already provided by package  * libnl-tiny2023-04-02
 * opkg_install_cmd: Cannot install package luci-app-openvpn.
```
the previous version openwrt v19.** working without any issue.

Ayushmanwebdeveloper, 2023-07-26:
You can try removing `libnl-tiny2023-04-02` first before installing `libnl-tiny2023-07-01`.
By using `opkg remove libnl-tiny2023-04-02` or by searching & removing from installed software 
[image]

henok400, 2023-07-26:
There is no any libnl-tiny install my router.
I installing frash openwrt and searching libnl-tiny*** not found
result issue continued.

Ayushmanwebdeveloper, 2023-07-26:
Maybe you can try this https://forum.openwrt.org/t/install-luci-app-openvpn-package/166228

jow-, 2023-07-26:
Duplicate of https://github.com/openwrt/openwrt/issues/13082

closed 2023-07-26 by jow-