openwrt/luci #6494: Luci-mod-status : port status doesn't display all eth ports (x86/64)

badulesia, 2023-08-01:
## Steps to reproduce:
Use an x86/64 mini PC with 4 eth ports
go to: Status → Overview → Port status

## Actual behavior:
Only eth0 and eth1 are displayed.

## Expected behavior:
4 eth ports should be displayed (eth 0 to eth3)

## Additional Information:

Brand : Qotom
Model | INTEL Corporation Q3XXG4-P
Architecture | Intel(R) Core(TM) i3-5005U CPU @ 2.00GHz
Target Platform | x86/64
Firmware Version | OpenWrt SNAPSHOT r23659-e6b72de703 / LuCI Master git-23.158.78004-23a246e
Kernel Version | 5.15.123

Luci-mod-status  | git-23.210.72211-00a941e

NICs are Intel I211 gigabit network

jow-, 2023-08-03:
Known issue, it only enumerates ports in /etc/board.json, add them there and they will appear.

badulesia, 2023-08-03:
Understood.
So I assume that /etc/board.json is fed by default with only eth0 and eth1, the strict minimum for a router to operate.

closed 2023-08-03 by badulesia

badulesia, 2023-08-04:
I manage to add extra eth ports to /etc/board.json

```
"network": {
		"lan": {
			"ports": ["eth0", "eth1", "eth2"],
			"protocol": "static"
		},
		"wan": {
			"device": "eth3",
			"protocol": "dhcp"
		}
	}
```