openwrt/luci #6495: luci: mesh_fwding must be written to mesh11sd if the config exists instead of wireless

Sophronia-AK, 2023-08-02:
In the wireless interface configuration section, the mesh forwarding config is available. 
This configuration must be mapped to mesh11sd uci config if the mesh11sd uci is available oand only be written to wireless if mesh11sd uci does not exist.

bluewavenet, 2023-08-02:
@Sophronia-AK 

> This configuration must be mapped to mesh11sd uci config 

No, that is not necessary because mesh11sd overrides whatever mesh parameters that may be set in the wireless config.

If the mesh11sd daemon is running in its default management mode, then if you want to set mesh forwarding to "0" then do it in the mesh11sd config. Note this will be dynamic and take effect immediately without having to restart the wireless. This means you can turn it off and on at will, keeping everything up (this goes for all supported mesh parameters). Don't do a `uci commit mesh11sd` unless you want the change to survive a reboot.

closed 2023-08-03 by jow-

Sophronia-AK, 2023-08-10:
@bluewavenet  Thank you it works !