openwrt/luci #6501: luci-mod-admin-mini: Backup archive does not contain installed files

jmigual, 2023-08-05:
## Steps to reproduce:

1. Go to System → Backup / Flash Firmware
2. On the Backup section, click on "Generate archive"
3. Extract the archive

## Actual behavior:

The file `etc/backup/installed_packages.txt` is not in the backup

## Expected behavior:
The file `etc/backup/installed_packages.txt` is not in the backup

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS=''
```

I've been browsing the code and, if I understood it right, the part choosing the files to backup is https://github.com/openwrt/luci/blob/0ea74956c9e7ae415cf6ef11d810888c422db309/modules/luci-mod-admin-mini/luasrc/controller/mini/system.lua#L16-L55

And specially I think that line 19 should be replaced by: 

```lua
local backup_cmd  = "sysupgrade -k -b - 2>/dev/null"
```

As this already generates a `.tar.gz` file that gets sent to the command line.

hnyman, 2023-08-05:
LuCI mini has been declared "BROKEN" (=faulty, unsupported) years ago, in January 2015.

The whole mod should probably be removed.  (or alternatively, modernised).

cc @jow-

jmigual, 2023-08-05:
Ah so then then I searched the code in the wrong place. What is the mod in charge of this place?

[image]

hnyman, 2023-08-05:
Well, in master
https://github.com/openwrt/luci/blob/master/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js

jmigual, 2023-08-05:
Okay, then maybe I made a mistake pointing to LuCi mini because I just downloaded the file using the "Generate archive" button and it still doesn't have the expected `installed_packages.txt`. 

Could it be that LuCi mini is still being used in this call?

But the generate archive button is still handled by luci mini right? Who is receiving this POST call?

https://github.com/openwrt/luci/blob/0ea74956c9e7ae415cf6ef11d810888c422db309/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js#L80-L91

jow-, 2023-08-05:
Why is the backup archive expected to contain a "installed_packages.txt" ? It never did, and it's nowhere advertised in the ui?

jmigual, 2023-08-06:
Ah, my apologies then. Could this issue be a request to add a check box to have the `installed_packages.txt` also in the backups?

hnyman, 2023-08-06:
> is the backup archive expected to contain a "installed_packages.txt" ? It never did, and it's nowhere advertised in the ui?

Sysupgrade actually supports including that with "-k" for a backup:

```
root@router5:~# sysupgrade -k -b koe.tar.gz
Sun Aug  6 21:20:24 EEST 2023 upgrade: Saving config files...

root@router5:~# tar -tzf koe.tar.gz  | grep inst
etc/backup/installed_packages.txt
```

And since commit 0f4f6e8 in 2021 LuCI does support the "-k" flag for the sysupgrade/flashing itself 
[image]

But that is not enabled for the plain backup. Perhaps we should extend the functionality there.

hnyman, 2023-08-06:
as we are using the cgi-io package for the backup (as cgi-backup), providing the "-k" would require passing it to the cgi-io package
https://git.openwrt.org/?p=project/cgi-io.git;a=blob;f=main.c;h=8ca4c046d96eefe88aa8221e93814d0de7a0994c;hb=HEAD#l614
in the call from
https://github.com/openwrt/luci/blob/master/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js#L80-L85

Probably as a new field? (in addition to sessionid)

So, providing that functionality is somewhat more complicated than just editing the LuCI page.

raenye, 2023-10-11:
> Probably as a new field? (in addition to sessionid)

Something like this? (place at `package/feeds/packages/cgi-io/patches/include_packages_txt.patch`).
It compiles but I haven't tested it (front end code is needed).

[include_packages_txt.patch](https://github.com/openwrt/luci/files/12863094/include_packages_txt.patch)