openwrt/luci #6520: luci-proto-wireguard "Generate configuration" Information not complete

ignis-draco, 2023-08-15:
## Steps to reproduce:
1. install luci-proto-wireguard
2. Create a WireGuard VPN interfaces
3. setup Generall Settings 
4. go to Advanced Setting -> Use Custom DNS Servers 
5. go to peer -> edit or add Peer 
6. Configure peer: "Allowed IPs" and "Endpoint Port"
7. klich on "Generate configuration"

## Actual behavior:

1. The Set Ip Adress is not in der Interface section 
2. If you scan the QR code, no connection can be established because no IP address is set. 
3. The DNS server configured in the interface is also not entered. 

Configuiration created:
`
[Interface]
PrivateKey = 111111111111111111111111111111111111
ListenPort = 51880

[Peer]
PublicKey = 22222222222222222222222222222222222222
# PresharedKey not used
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = home.example.com:51880
PersistentKeepAlive = 25
`

## Expected behavior:

Expected configuiration
`[Interface]
PrivateKey = 111111111111111111111111111111111111
**Address = 192.168.9.2/24**
**DNS = 8.8.8.8**

[Peer]
PublicKey = 22222222222222222222222222222222222222
# PresharedKey not used
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = home.example.com:51880
PersistentKeepAlive = 25`

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='bcm27xx/bcm2711'
DISTRIB_ARCH='aarch64_cortex-a72'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS=''

```

jow-, 2023-08-21:
I don't think it makes sense to have WireGuard peers use the same upstream DNS server as the router itself. For example the NS might not be reachable via the VPN.

ignis-draco, 2023-08-21:
> I don't think it makes sense to have WireGuard peers use the same upstream DNS server as the router itself. For example the NS might not be reachable via the VPN.

If you use a DNS addblocker, it is very beneficial if the peer also uses this DNS server.

> Where is the `home.example.com` supposed to come from?

I put that there because I didn't want to post my real domain. 

> Missing DNS and Address options are known and already tracked in
> #6351
> and
> #6352

Ah sorry I had not found the two entries in my search.

ignis-draco, 2023-08-21:
That had confused me a bit at first. I set my dyndns entry via Corn and Curl (I had already done this before I switched to OpenWRT). 
But after reading the documentation a bit, it was clear to me that I need the dyndns plugin. After I set it up appropriately, it worked without a problem. If you really want to improve something, you can save the entry you made under "Connection endpoint" -> coustom. But I think the variant with the Dyndns entry is better because you only have one place where you have to change something.

systemcrash, 2023-08-24:
> I don't think it makes sense to have WireGuard peers use the same upstream DNS server as the router itself. For example the NS might not be reachable via the VPN.

True. But it conversely also might be reachable. I think using such an IP would be a sensible default, though not a panacea. A checkbox option for the dns server would be a choice the admin knows.

systemcrash, 2023-12-04:
Ah, I see: you meant 'use the same upstream DNS server as the router itself uses'. Clear.

sshambar, 2024-01-18:
BTW, I tried using the "Generate configuration" popup window from one of the Peer configs, and as my public address is not the router address (it's forwarded to the router from the gateway), I tried adding a "--custom--" entry for the "Connection endpoint" value...

However, after adding the new DNS value, neither the QR code nor the config were updated; in fact, editing any of the values in the modal popup didn't update the QR/config.  There's probably a missing event trigger there someplace :)

jow-, 2024-03-04:
DNS and address settings are implemented now, the failing refresh has been addressed in #6921 / e28f9a0491f5f50a70bbefb1720ff8241f456f83

closed 2024-03-04 by jow-