openwrt/luci #6522: `luci-mod-network` Route silently fails if host address left in target network.

valschmidt, 2023-08-15:
## Steps to reproduce:

1. go to: Network → Routing 
2. Click Add
3. Configure a route with the following: Interface: 'WAN' (anything will do), Route type: unicast, Target: 10.0.0.4/24, Gateway: 10.0.0.1
4. Save
5. Save and Apply
6. Observe no warning or error that the Target network was not specified correctly, and the route was not set.

## Actual behavior:
1. Login to router as root. 
2. Observe the /etc/config/network contains the route for the 10.0.0.4/24 network.
3. Execute `ip r` and observe that the route for 10.0.0.0/24 does not exist in the routing table. `uci` has silently failed to configure it. 

Note that reconfiguring the route above with a Target network of 10.0.0.0/24 will result in the route being properly set by `uci`. One must clear the host bits of the network address.

## Expected behavior:
The `luci` interface should not allow one to configure a Target network with the bits beyond the subnet non-zero if `uci` is unable to configure a route without them cleared. Alternatively, if `uci` is unable to successfully configure the route, `luci` should detect this and warn the user rather than silently fail. 

## Additional Information:
OpenWrt 22.03.3
```

valschmidt, 2023-08-21:
I totally agree. I'm not familiar with the backend, is netifd what executes the uci commands to apply the settings?

jow-, 2024-12-04:
This has been solved with https://github.com/openwrt/netifd/commit/76eb342c397a90feb77b54be2706fa9ff61637f1

closed 2024-12-04 by jow-