openwrt/luci #6528: luci-lua-runtime/luci-base: luci menu cache issue [23.05]

jjm2473, 2023-08-21:
After wifi module auto configured, menu not shown "Wireless". Same issue appears on Docker, if I open luci on browser on boot, Docker maybe not start on that time, menu will not show Docker menu even after docker started.

`rm -f /tmp/luci-indexcache*` drop the cache resolv that.

jjm2473, 2023-08-21:
In the previous version, for the lua controller, the lua code was cached, not the menu.

jow-, 2023-08-22:
I'm afraid that the caching is something I won't roll back, constantly rebuilding the menu and synchronizing it with the client is just too much work, but maybe we can figure out a smarter way for cache invalidation.

jjm2473, 2023-08-22:
I use this patch to force outdate cache after 5 seconds https://github.com/jjm2473/luci/commit/a430de5b8a27bd04ca00640c111eb0768095a8eb

May still cause performance impact on low-performance platforms.

jjm2473, 2023-08-22:
I suggest lua controller cache code, and the json menu add dependencies to the calculation of cache hash

wryun, 2024-04-12:
See https://github.com/openwrt/luci/issues/6422