openwrt/luci #6530: luci-lua-runtime: file upload broken [23.05]

jjm2473, 2023-08-21:
I have a file upload controller:

```lua
function index()
    entry({"admin", "store", "upload"}, post("store_upload"))
end

function store_upload()
    local fd
    local path
    local finished = false
    local tmpdir = "/tmp/is-root/tmp"
    luci.http.setfilehandler(
        function(meta, chunk, eof)
            if not fd then
                path = tmpdir .. "/" .. meta.file
                nixio.fs.mkdirr(tmpdir)
                fd = io.open(path, "w")
            end
            if chunk then
                fd:write(chunk)
            end
            if eof then
                fd:close()
                finished = true
            end
        end
    )
    local code, out, err
    out = ""
    if finished then
        if string.lower(string.sub(path, -4, -1)) == ".run" then
            code, out, err = _action("sh", "-c", "ls -l \"%s\"; md5sum \"%s\" 2>/dev/null; chmod 755 \"%s\" && \"%s\"; RET=$?; rm -f \"%s\"; exit $RET" %{ path, path, path, path, path })
        else
            code, out, err = _action("sh", "-c", "opkg install \"%s\"; RET=$?; rm -f \"%s\"; exit $RET" %{ path, path })
        end
    else
        code = 500
        err = "upload failed!"
    end
    --nixio.fs.unlink(path)
    local ret = {
        code = code,
        stdout = out,
        stderr = err
    }
    luci.http.prepare_content("application/json")
    luci.http.write_json(ret)
end
```

It was worked on 22.03.
But a broken file I get on 23.05.

jow-, 2023-08-21:
Will look into it. But please be aware that the above code is a direct shell injection vulnerability. The `meta.file` value is controlled by the client and part of the `Content-Disposition` HTTP header. You append that user supplied value to your temp directory prefix as `path` variable and then use `path` as-is interpolated into a shell command. At the very least, wrap `path` into a `luci.util.shellquote()` or even better, run a pattern match against `meta.file` and only accept sane values.

jjm2473, 2023-08-21:
> But please be aware that the above code is a direct shell injection vulnerability.

It's designed, for self-executable installer: user upload a *.run file, and I will execute it.

> I believe the issue is caused because nothing actually reads the HTTP parameters, so the upload HTTP message body is never parsed

Does the POST request always try `test_post_security`? If that do, `http.formvalue("token")` should be called already.

jjm2473, 2023-08-21:
I add `luci.http.formvalue()` and some logs:
```lua
function store_upload()
    local fd
    local path
    local finished = false
    local tmpdir = "/tmp/is-root/tmp"
    local size = 0
    nixio.syslog("crit", "setfilehandler")
    luci.http.setfilehandler(
        function(meta, chunk, eof)
            if finished then
                nixio.syslog("crit", "filehandler call after eof")
                return
            end
            if not fd then
                path = tmpdir .. "/" .. meta.file
                nixio.fs.mkdirr(tmpdir)
                fd = io.open(path, "w")
            end
            if chunk then
                size = #chunk + size
                fd:write(chunk)
            end
            if eof then
                fd:close()
                finished = true
                nixio.syslog("crit", "filehandler write %d bytes" % { size })
            end
        end
    )
    luci.http.formvalue()

    local code, out, err
    out = ""
    if finished then
        if string.lower(string.sub(path, -4, -1)) == ".run" then
            code, out, err = _action("sh", "-c", "ls -l \"%s\"; md5sum \"%s\" 2>/dev/null; chmod 755 \"%s\" && \"%s\"; RET=$?; rm -f \"%s\"; exit $RET" %{ path, path, path, path, path })
        else
            code, out, err = _action("sh", "-c", "opkg install \"%s\"; RET=$?; rm -f \"%s\"; exit $RET" %{ path, path })
        end
    else
        code = 500
        err = "upload failed!"
    end
    --nixio.fs.unlink(path)
    local ret = {
        code = code,
        stdout = out,
        stderr = err
    }
    luci.http.prepare_content("application/json")
    luci.http.write_json(ret)
end
```

Upload two files both around 15MB, get a broken file (only get head of file), and this log:

```
Mon Aug 21 17:29:42 2023 user.crit : setfilehandler
Mon Aug 21 17:29:42 2023 user.crit : filehandler write 81920 bytes
Mon Aug 21 17:31:32 2023 user.crit : setfilehandler
Mon Aug 21 17:31:32 2023 user.crit : filehandler write 668672 bytes
```

jow-, 2023-08-21:
Ah I see, I was not testing it with a `post()` action but a `call()` one. Will retry with large uploads later.

jjm2473, 2023-08-21:
@jow- `form()` has the same issue

commit b4594b3d (2023-08-21): luci-base: http.uc: fix eof detection in temporary upload files

Binary string comparisons in ucode are currently unsafe, so use the
`length()` function to determine the just read data chunk size in
order to test for end of file.

Fixes: #6530
commit f03bb9a1 (2023-08-21): luci-base: http.uc: fix eof detection in temporary upload files

Binary string comparisons in ucode are currently unsafe, so use the
`length()` function to determine the just read data chunk size in
order to test for end of file.

Fixes: #6530

(cherry picked from commit b4594b3d111171510d2581e9f0ce30bee78bd214)
closed 2023-08-21 by jow-