openwrt/luci #6534: LUCI: Port Status on System Status Page Doesn't Show "Ethernet Gadget" Port

Manley1995, 2023-08-21:
## Steps to reproduce:
(Note - this example is from an APU4 device, with physical ports eth0-eth3)
1. Configure Raspberry Pi0 as "ethernet gadget"; connect Pi0's USB data port to router's USB port
2. Install kmod-usb-net-cdc-ether in OpenWRT
3. Confirm that Pi0 shows up as ethernet device on Network/Interfaces page, Devices tab (e.g., eth4)
4. Add Pi0 "gadget" (eth4) to br-lan
5. Confirm traffic is passing to/from Pi0 (e.g., ssh into Pi0)
6. Go to Status/Overview page. Look at Port Status indicators. See (in this example) eth0 - eth3.
7. Notice that there is no indicator for eth4 (the Pi0 ethernet gadget)

## Actual behavior:
Port status indicators show (in this example) for eth0-eth3, but no indicator shows for eth4

## Expected behavior:
A port status indicator should show for the ethernet gadget port (eth4), indicating the zone(s) it belongs to (just like the indicators for the physical ethernet ports)

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc3'
DISTRIB_REVISION='r23389-5deed175a5'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc3 r23389-5deed175a5'
DISTRIB_TAINTS=''

jow-, 2023-08-22:
It is really only meant to display builtin devices and not random optional hardware attachments. Main focus was integrated router systems, not dyi platforms. You can add "eth4" to /etc/board.json though to treat it as builtin hardware port.

closed 2023-08-22 by jow-

Manley1995, 2023-08-22:
Thanks.  Adding "eth4" to /etc/board.json works.  Since eth4 is automatically detected and shows up under Network/Interfaces (Devices tab), and shows up in the little popup text boxes in the Network/Interfaces page, I figured it would also be included in other displays of ethernet devices (like the main System/Status page).  Regardless, I appreciate the workaround.

dongc123, 2023-08-24:
> Hello, I tried to add the eth3 port in /etc/board.json and after restarting openwrt, luci still did not display the eth3 port.

jow-, 2023-08-24:
```json
{
    "network": {
        "lan": {
            "protocol": "static",
            "ports": [ "eth0", "eth2", "eth3" ]
        },
        "wan": {
            "protocol": "dhcp",
            "device": "eth1"
        }
    }
}
```

dongc123, 2023-08-24:
> thank you solved the problem

[image]

dongc123, 2023-08-24:
> ```json
> {
>     "network": {
>         "lan": {
>             "protocol": "static",
>             "ports": [ "eth0", "eth2", "eth3" ]
>         },
>         "wan": {
>             "protocol": "dhcp",
>             "device": "eth1"
>         }
>     }
> }
> ```

{
	"model": {
		"id": "vmware-inc-vmware20-1",
		"name": "VMware, Inc. VMware20,1"
	},
	"network": {
		"lan": {
			"ports": ["eth0","eth2","eth1"]
		}
	}
}
Writing luci in this way shows the same result, is it possible?

jow-, 2023-08-24:
You likely need to log out and in again and possible force a browser cache clean.