openwrt/luci #6538: OpenWrt 23.05.0-rc3 reports a luci bug

dongc123, 2023-08-24:
The luci port status has no Chinese display, and there are 4 ports that only display eth0 and those behind eth1 are not displayed.
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc3'
DISTRIB_REVISION='r23389-5deed175a5'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc3 r23389-5deed175a5'
DISTRIB_TAINTS=''
```
![微信截图_20230824205406](https://github.com/openwrt/luci/assets/109882914/42a6bd02-836a-4519-8b8b-bb61644565a9)

dongc123, 2023-08-24:
[image]

jow-, 2023-08-24:
Duplicate of https://github.com/openwrt/luci/issues/6534.

On x86 you need to add additional ethernet interfaces to `/etc/board.json` - this will not be solved in LuCI but should eventually happen automatically in OpenWrt.

Missing chinese translation will appear once translators catch up, afair it simply hasn't been translated yet.

jow-, 2023-08-24:
Chinese translations pushed with 9f3e3985c685a37ec83e6a10bbc694ba8f0dc3cb, will appear in the package repos within the next few days.

closed 2023-08-24 by jow-

dongc123, 2023-08-24:
> #6534的重复。
> 
> 在 x86 上，您需要添加额外的以太网接口`/etc/board.json`- 这不会在 LuCI 中解决，但最终应该在 OpenWrt 中自动发生。
> 
> 一旦翻译者赶上，就会出现缺少的中文翻译，而且它根本还没有翻译。

Hello, I tried to add the eth3 port in /etc/board.json and after restarting openwrt, luci still did not display the eth3 port.
[image]