openwrt/luci #6540: Port Status Overview - Feedback

shauno100, 2023-08-25:
Hi, this is not an issue but more feedback. I am really liking the new port status overview in OpenWrt 23.05 RC3 on Xiaomi AX6000 router but would it be more logical to have the red underline for ports that are not connected along with the term "no link" instead of using red to differentiate WAN from the LAN ports where the LAN ports have a green underline regardless of whether they are connected or not?

Just didn't make much sense to me that's all to differentiate on port type as you'd expect green for connected ports regardless of type and red for disconnected ports etc.

[image]

ErwanMAS, 2023-08-26:
color indicate firewall zone

shauno100, 2023-08-27:
Fair enough, how I see it though and maybe it's just me, green means connected and red means disconnected. Logically this makes sense rather then colour coding port status by firewall zone especially when those colours are red and green. I feel several people would pay more attention to the colour at first glance when checking if a port is connected.

ErwanMAS, 2023-08-27:
But i agree with you , that this layout is not perfect and bring confusion .

FYI in case you have some skills to submit a PR , the layout is build here

 https://github.com/openwrt/luci/blob/ed059e4cd3ee9b15aabcc974b4e66b0090b920ae/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/29_ports.js#L342-L374

shauno100, 2023-08-28:
Unfortunately i don't have the skills, just thought that maybe whoever wrote the code might see this issue and consider it that's all.

shauno100, 2023-08-29:
In addition would having live port statistics for up/down traffic be possible/better to display (e.g. the down KBps and up etc at the current point in time the ports are being viewed in conjunction with the auto refresh of the webpage) rather than the accumulated total?

aleks-mariusz, 2023-09-21:
> In addition would having live port statistics for up/down traffic be possible/better to display (e.g. the down KBps and up etc at the current point in time the ports are being viewed in conjunction with the auto refresh of the webpage) rather than the accumulated total?

I was thinking this was actually a really clever idea.. but OpenWRT already has the ability to show you live traffic stats on a per-interface basis (Status -> Realtime graphs -> Traffic).. what would be also cool then building atop of @shauno100 's idea is maybe in-addition to their idea (or instead of it if there's some reason it's not feasible), if clicking on that link could take you to the right graph also/instead?

aleks-mariusz, 2023-09-21:
> But i agree with you , that this layout is not perfect and bring confusion .

What if we made the bar not have any color if the link is down?  The fact it's still left green intuitively suggests to the user it's all good/operational (even if the color has a meaning about the zone itself).. or for extra credit, make this configurable even somehow? since in my experience, there'll always be someone that will find a reason to disagree with things changing in any way :-)

shauno100, 2023-11-26:
> > But i agree with you , that this layout is not perfect and bring confusion .
> 
> What if we made the bar not have _any_ colors if the link is down? The fact it's still left green intuitively suggests to the user it's all good/operational (even if the color chosen has a meaning about the zone itself).. or for extra credit, make this configurable even somehow? since in my experience, there'll always be someone that will find a reason to disagree with things changing in any way :-)

This sounds good to me, the customization of the colors will probably never happen but in recent OpenWrt 23.05.2 i can see the LAN ports that are down (not connected) are a dull green color whereas the connected LAN port is a brighter green. WAN port still is red as per the firewall zone color scheme.

Anyway just my opinion but i still reckon a red colour for down ports (regardless if LAN or WAN) and a green colour for up and active ports (LAN or WAN) would make most sense to me. Only my opinion as mentioned and the majority may not agree with it but thought i'd make it known in case the LUCI developers are reading the issue comments,

lienmeat, 2025-06-04:
New openwrt user here, just getting the openwrt one set up.  I have to agree with this feedback. I spent several minutes (maybe 30?) wondering why WAN was not connected/didn't have internet despite a lot of evidence otherwise, due to the color, until I decided the UI was wrong and I didn't care.  Just now, I finally realize the color is tied to the firewall zone and not the port's "status".  And my very next thought is "how do I make WAN blue or something".  Red is a very bad color for something if it could be misconstrued to mean "off" or something in a UI.  I would urge devs (or I might get frustrated enough to make a PR) to take this feedback seriously, and at minimum, change the colors to be perhaps blue and purple or anything else that doesn't to many people mean "on" or "off", but better, make it configurable.

lienmeat, 2025-06-04:
easy workaround for me, I just sshed in, vim /www/luci-static/resources/firewall.js
change this function to whatever colors you want for lan and wan:

`function getColorForName(forName) {
	if (forName == null)
		return '#eeeeee';
	else if (forName == 'lan')
		return '#90f090';
	else if (forName == 'wan')
		return '#f09090';

	return random.derive_color(forName);
}`