openwrt/luci #6543: luci-app-firewall: "Forward" drop-down has no effect, but is not grayed out, if "Allow forward to destination zones" is set

alexhenrie, 2023-08-28:
## Steps to reproduce:
1. go to: Network → Firewall
2. look at the default settings for the lan zone
[image]
3. change the "Forward" drop-down from "accept" to "reject" and save

## Actual behavior:
Nothing changes. No packets are rejected that were not rejected before.

## Expected behavior:
_Something_ should stop working.

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc3'
DISTRIB_REVISION='r23389-5deed175a5'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc3 r23389-5deed175a5'
DISTRIB_TAINTS=''
```

The reason why nothing changes is because in the Edit dialog, "Allow forward to destination zones" is set:

[image]

So the firewall rules look like this (excerpts from `nft list ruleset`):

```
chain forward_lan {
        jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
        jump reject_to_lan
}
```

```
chain accept_to_wan {
        meta nfproto ipv4 oifname "eth0.2" ct state invalid counter packets 113 bytes 6217 drop comment "!fw4: Prevent NAT leakage"
        oifname "eth0.2" counter packets 4562 bytes 626946 accept comment "!fw4: accept wan IPv4/IPv6 traffic"
}
```

```
chain reject_to_lan {
        oifname "br-lan" counter packets 0 bytes 0 jump handle_reject comment "!fw4: reject lan IPv4/IPv6 traffic"
}
```

The `jump accept_to_wan` rule (created by "Allow forward to destination zones") completely overrides the `jump reject_to_lan` rule. Either that's a bug and there should be a `return` rule at the end of the `accept_to_wan` chain, or the "Forward" drop-down should be grayed out when "Allow forward to destination zones" is set, to clearly communicate that it no longer has any effect.

To be fair, the current behavior is not so bad if you leave everything on the default settings, but it gets really confusing when defining custom zones and VLANs.

jow-, 2023-08-28:
That dropdown controls forwarding policy for traffic entering and leaving the same zone, so either among muliple interfaces covered by the zone or forwarded through the interface. It does not control traffic towards other zones.

closed 2023-08-28 by jow-

alexhenrie, 2023-08-28:
Thanks for the explanation Jo. Is there any way we could make the user interface less confusing here? "Forward: reject" is right next to the label **lan ⇒ wan**, but if I understand you correctly, it actually applies to **lan ⟺ lan** traffic. Even adding an extra sentence to the Edit dialog to explain the difference would be very helpful.