openwrt/luci #6551: Breadcrumb: Administration button missing when not logged in (In the snapshot version)

Ayushmanwebdeveloper, 2023-08-31:
This issue does not exist in previous versions; it is only present in the snapshot version. While migrating the luci-app-olsr app, we decided to expose the status views to the public, which means they can be accessed without authentication. This worked for previous openwrt versions; however, in the snapshot version, the admin button is not visible without login, preventing the user from returning to admin. 

## Steps to reproduce:

1. Install luci-app-olsr in the latest snapshot version of OpenWrt
2. Just visit the `/cgi-bin/luci/` 
4. The administration button is missing.
[image]
5. Visit `/cgi-bin/luci/admin`
6. Login
7. Button can be seen now in both admin & olsr status views.
[image]

## Actual behavior:
The admin button is not visible without login in the snapshot version.

## Expected behavior:
The admin button should be visible without login.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r23797-ff2379662c'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r23797-ff2379662c'
DISTRIB_TAINTS=''
```

andibraeu, 2023-08-31:
@jow- do you have an idea?

/CC @thuehn

jow-, 2023-08-31:
Yeah, I do. Fix is nontrivial though and will take a while.