openwrt/luci #6555: App appears to have an upper limit of config files it can display, before displays empty list

itsthejb, 2023-09-02:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. Add a long list of configurations via `.opvn` files in `/etc/config/openvn`:

```
config openvpn 'MyVPM'
	option config '/etc/openvpn/somevpn.opvn'
```

A long-ish list: probably more than ~25 (not sure the actual number). Large number, but not _huge_

2. Go to the VPN page in Luci

## Actual behavior:

The VPN config list will be _empty_

## Expected behavior:

Complete list is shown.

In order to get a list, must remove entries until the config list is small: ~25 or fewer

## Additional Information:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc2'
DISTRIB_REVISION='r23228-cd17d8df2a'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc2 r23228-cd17d8df2a'
DISTRIB_TAINTS=''
```

jow-, 2023-09-05:
Cannot reproduce.

[image]
[image]

closed 2023-09-05 by jow-

itsthejb, 2023-09-05:
Ok, thanks for checking. Yes, you're correct - I found that my list ended up including my names with a period which are rejected. Now fixed. Thanks!