openwrt/luci #6557: luci: cannot change timezone as Language and Style -> Design requires non-empty value; no choices

nokolade, 2023-09-03:
## Steps to reproduce:

1. go to: System → General Settings
2. change timezone
3. Tab Language and Style shows (1) issue - tooltip: 1 invalid field(s)
4. On Tab Language and Style->Design is framed red; expecting: non-empty value yet no choices given

## Actual behavior:
Tab Language ans Style->Design is framed red; expecting: non-empty value yet no choices given.
It is not possible to save the changed timezone.

## Expected behavior:
It is possible to save the changed timezone without being bothered with Design selection.

## Additional Information:
I've briefly tried
# uci set system.@system[0].timezone=CET-1CEST,M3.5.0,M10.5.0/3
# uci commit
but luci still stated UTC (admittedly I did not restart/reboot so maybe the command line change just did not propagate into the UI. To test uci, I've taken the string "CET-1CEST,M3.5.0,M10.5.0/3" from my OpenWRT22.03 router.

OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc3'
DISTRIB_REVISION='r23389-5deed175a5'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc3 r23389-5deed175a5'
DISTRIB_TAINTS=''
```

nokolade, 2023-09-03:
PS: just found the issue does NOT exist on my WiFi router. Here, the "Design" drop down lists Bootstrap etc.
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc3'
DISTRIB_REVISION='r23389-5deed175a5'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc3 r23389-5deed175a5'
DISTRIB_TAINTS=''

stokito, 2023-09-05:
For some reason you have empty `modularbase` option in /etc/config/luci.
The design option is required (not optional = true):
https://github.com/openwrt/luci/blob/master/modules/luci-mod-system/htdocs/luci-static/resources/view/system/system.js#L237

The part of Luci didn't changed for a while so it's something wrong with your config.

jow-, 2023-09-05:
Your /etc/config/luci is broken, it should contain a number of theme entries which are added when a `luci-theme-*` package is installed. 

To restore the default theme entry, run the following commands:
```
uci set luci.themes=internal
uci set luci.themes.Bootstrap='/luci-static/bootstrap'
uci set luci.themes.BootstrapDark='/luci-static/bootstrap-dark'
uci set luci.themes.BootstrapLight='/luci-static/bootstrap-light'
uci commit luci
```

Alternatively do a `opkg remove luci-theme-bootstrap; opkg install luci-theme-bootstrap`.

Closing this as it is caused by a local broken config (introduced by config restore?)

closed 2023-09-05 by jow-

nokolade, 2023-09-09:
Thank you @jow- , this helped!
Seems that blindly moving luci-opkg to luci in /etc/config after updating the packages was not a smart idea.
Will be more careful in future.