openwrt/luci #6569: Unable to execute opkg install command: SyntaxError: Unexpected end of JSON input

odkrys, 2023-09-07:
```
Unable to execute opkg install command: SyntaxError: Unexpected end of JSON input
```
I have a question.
When I install samba4-libs on a fortigate 50e with 20MB of free space, an error message as in the title appears.

I know that samba4-libs is compressed and is about 8MB, but when decompressed it is over 20MB.
However, the overlay partition is a compressed space, so when samba4-libs is actually installed through CLI, it uses 13.2MB of space and there is still 7MB of free space on the device.

When I install samba4-libs from Luci, I get an error, but if I actually check it via "ps", I can see that the package is still being installed.

That's why the error message feels like a false alarm. Is this something that can be fixed?

hnyman, 2023-09-07:
Which OpenWrt version?

That JSON error from opkg "Unexpected end of JSON input" sounds like a really long dependency line cut in middle, or something like that. I have a feeling that @jow- fixed something on that front a while ago.

odkrys, 2023-09-07:
today's snapshot

root@OpenWrt:/# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r23885-703d667a0c'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r23885-703d667a0c'
DISTRIB_TAINTS=''

jow-, 2023-09-07:
The invocation of `/usr/libexec/opkg-call` via `cgi-exec` likely fails, probably because a timeout was reached. I don't think that can be fixed LuCI side.

odkrys, 2023-09-08:
> probably because a timeout was reached

Am I correct in saying that this is what can happen when installing a large amount of files on a slow cpu?
That's sad, I understand.

closed 2023-09-08 by odkrys