openwrt/luci #6580: [23.05.0-rc3] ASUS RT-AX53U | Status > Overview "InternalError too much recursion"

SeimusS, 2023-09-20:
# Issue:
On the latest 23.05.0-rc3 the Status > Overview wont load it gives "InternalError too much recursion". Tried as well 23.05.0-rc2 this issue was not seen on this release.

I checked with other user that is running same Router and same OS version/same snapshot and he doesnt have this issue. The main difference is that I am using VLANs and several VAPs

Clearing cache and cookies doesnt help, I tried as well 3 different browsers still the same.

To be more precise on the issue, only the status/overview doesnt load "/cgi-bin/luci/admin/status/overview"
[image]

## Steps to reproduce:
Go to Status > Overview. Must have VLANs.

## Actual behavior:
Cant load Overview in luci

## Expected behavior:
Status > Overview should show general overview

## Additional Information:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0-rc3'
DISTRIB_REVISION='r23389-5deed175a5'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0-rc3 r23389-5deed175a5'
DISTRIB_TAINTS=''

## Logs

**Output from browser:**

RPCError: RPC call to luci/getBuiltinEthernetPorts failed with error -32002: Access denied
at handleCallReply (https://10.0.0.152/luci-static/resources/rpc.js?v=git-23.236.53405-fc638c8:11:3)
raise https://10.0.0.152/luci-static/resources/luci.js?v=git-23.236.53405-fc638c8:155
handleCallReply https://10.0.0.152/luci-static/resources/rpc.js?v=git-23.236.53405-fc638c8:11
[luci.js:161:9](https://10.0.0.152/luci-static/resources/luci.js?v=git-23.236.53405-fc638c8)
InternalError: too much recursion
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
[luci.js:161:9](https://10.0.0.152/luci-static/resources/luci.js?v=git-23.236.53405-fc638c8)
Uncaught (in promise) InternalError: too much recursion
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39

jow-, 2023-09-25:
Please provide your complete /etc/config/network and /etc/board.json files.

SeimusS, 2023-09-25:
As requested by Jo:

```
root@ASus_OpenWRT_AP01:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'
        option ula_prefix 'fdb6:b19f:016d::/48'

config device
        option type 'bridge'
        option name 'VLAN'
        option bridge_empty '1'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'VLAN.10'
        list ports 'VLAN.50'
        list ports 'VLAN.101'
        list ports 'VLAN.201'
        list ports 'VLAN.999'
        option ipv6 '0'

config bridge-vlan
        option device 'VLAN'
        option vlan '10'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'VLAN.10:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '50'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'VLAN.50:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '101'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'VLAN.101:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '201'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'VLAN.201:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '999'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'VLAN.999:t'

config interface 'VLAN10_HOST'
        option proto 'none'
        option device 'VLAN.10'

config interface 'VLAN50_CONSOLE'
        option proto 'none'
        option device 'VLAN.50'

config interface 'VLAN101_WORK'
        option proto 'none'
        option device 'VLAN.101'

config interface 'VLAN201_IOT'
        option proto 'none'
        option device 'VLAN.201'
        option delegate '0'

config interface 'VLAN999_MGMT'
        option proto 'static'
        option device 'VLAN.999'
        option ipaddr '10.0.0.152'
        option netmask '255.255.255.0'
        option gateway '10.0.0.138'
        list dns '10.0.0.14'
        option delegate '0'
        option broadcast '10.0.0.255'

```

```
root@ASus_OpenWRT_AP01:~# cat /etc/board.json
{
        "model": {
                "id": "asus,rt-ax53u",
                "name": "ASUS RT-AX53U"
        },
        "led": {
                "usb": {
                        "name": "USB",
                        "sysfs": "blue:usb",
                        "type": "usbport",
                        "ports": [
                                "usb1-port2"
                        ]
                },
                "wlan2g": {
                        "name": "WiFi 2.4GHz",
                        "sysfs": "mt76-phy0",
                        "trigger": "phy0tpt"
                },
                "wlan5g": {
                        "name": "WiFi 5GHz",
                        "sysfs": "mt76-phy1",
                        "trigger": "phy1tpt"
                }
        },
        "network": {
                "lan": {
                        "ports": [
                                "lan1",
                                "lan2",
                                "lan3"
                        ],
                        "protocol": "static"
                },
                "wan": {
                        "device": "wan",
                        "protocol": "dhcp"
                }
        },
        "system": {
                "compat_version": "1.1"
        }
}
```

SeimusS, 2023-10-02:
Updated to 23.05.0-rc4 issue is still present.

Regards,
S.

raenye, 2023-10-10:
How can `VLAN.10`, etc., be ports of the `VLAN` bridge? this seems wrong...
AFAICT you should remove all these from the bridge and from the bridge-vlans. Having VLAN.10 in the interface should do the trick

SeimusS, 2023-10-10:
I dont think is wrong.

This is how are **TAGed VLANs** created. A bridge device is created and you enable Vlan bridge filtering with specific VLANs you need + allocate them to physical interfaces.

The indivudual L2 interfaces are created into the interface is a vlan attached. This is then reused in VAPs, so each VAP has a specific VLAN TAG.

Regards,
S.

raenye, 2023-10-10:
Please see examples here: https://github.com/openwrt/openwrt/issues/10219

(and here for swconfig and single-port DSA: https://openwrt.org/docs/guide-user/network/vlan/switch_configuration)

raenye, 2023-10-10:
Let me just note that the top message speaks of config that *should* work, and the extra bridge creation is the way to make it work for `brcmfmac`.
Also, if you're looking for VLANs for VAPs only, consider this:
https://forum.openwrt.org/t/individual-per-passphrase-wifi-vlans-using-wpa-psk-file-no-radius-required/161696/27

SeimusS, 2023-10-10:
So I went thru the links mentioned in your comment before. And you are right looks like mistake within the bridge-device. I did remove the VLAN listed ports from it which removed it as well from bridge-vlans. Now the Overview is working as should.

This as well now explains Why I was having the recursion error.

```
root@ASus_OpenWRT_AP01:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'
        option ula_prefix 'fdb6:b19f:016d::/48'

config device
        option type 'bridge'
        option name 'VLAN'
        option bridge_empty '1'
        option ipv6 '0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'

config bridge-vlan
        option device 'VLAN'
        option vlan '10'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '50'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '101'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '201'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'

config bridge-vlan
        option device 'VLAN'
        option vlan '999'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'

config interface 'VLAN10_HOST'
        option proto 'none'
        option device 'VLAN.10'

config interface 'VLAN50_CONSOLE'
        option proto 'none'
        option device 'VLAN.50'

config interface 'VLAN101_WORK'
        option proto 'none'
        option device 'VLAN.101'

config interface 'VLAN201_IOT'
        option proto 'none'
        option device 'VLAN.201'
        option delegate '0'

config interface 'VLAN999_MGMT'
        option proto 'static'
        option device 'VLAN.999'
        option ipaddr '10.0.0.152'
        option netmask '255.255.255.0'
        option gateway '10.0.0.138'
        list dns '10.0.0.14'
        option delegate '0'
        option broadcast '10.0.0.255'
```

> Let me just note that the top message speaks of config that _should_ work, and the extra bridge creation is the way to make it work for `brcmfmac`. Also, if you're looking for VLANs for VAPs only, [...]

I will have a look on this and see if I can simplyfy the VAP per VLAN deployment

Many thanks @raenye much appriciated.

Regards,
S.