openwrt/luci #6583: Bug: v23.05 rc2 DHCP Static IP, can not input fixed IP in android browser app

davidngrc, 2023-09-22:
I am in SamSung M51, with stock Android v12, chrome browser v117, also try some old webkit browser
After Login LUCI (in mobile or desktop view) -> Top Menu: Network -> Sub Menu: DHCP and DNS -> Sub-Top Menu: Static Leases -> Bottom: Add
Try Manually input HostName, MAC address and a custom IP v4 address.
You will find out, after you input above, and try kit the mobile keyboard on the custom IP, the IP is not inputted.
I also try other android on-screen keyboard, same issue.

Model	Xiaomi AX3600
Architecture	ARMv8 Processor rev 4
Target Platform	ipq807x/generic
Firmware Version	OpenWrt 23.05.0-rc2 r23228-cd17d8df2a / LuCI openwrt-23.05 branch git-23.118.79121-6fb185f
Kernel Version	5.15.118

davidngrc, 2023-09-24:
I just upgrade to v23.05 rc3 on the same device.
this issue still not fix.

jow-, 2023-10-19:
Cannot reproduce it in the chrome mobile device emulation and can't test on real hardware since I don't own any Android device. Maybe someone else can confirm/debug

systemcrash, 2023-11-13:
@davidngrc can you perform a screen recording demonstrating this?

davidngrc, 2023-11-15:
> @davidngrc can you perform a screen recording demonstrating this?

https://file.io/QQyMETgIg42W

systemcrash, 2023-11-15:
I tried the file at both links, and I cannot open it, nor does ffmpeg recognize it:
`[mov,mp4,m4a,3gp,3g2,mj2 @ 0x7fd486304100] moov atom not found
Screen_Recording_20231115_084126_Chrome.mp4: Invalid data found when processing input`

The data appears truncated at around 50.4mb within the file.

systemcrash, 2023-11-15:
OK, I see it happening. 

My GUI is only uncooperative when I don't already have a name there. Otherwise it works. Does mobile chrome have a developer pane which shows perhaps what is going on?

I might suggest trying a different keyboard.

davidngrc, 2023-11-15:
I try 3 other keyboard already, this is the samsung stock, just reset, did not install anything else.
so there is an issue here.

systemcrash, 2023-11-15:
Do you still get the problem if you put the cursor into each of the fields at least once? Name, MAC, IP?

davidngrc, 2023-11-16:
@systemcrash  still same issue
Here is new test video on another phone (not reset) with kiwi browser and 3 keyboard.
This browser include developer tool, I also shown it in the video

https://we.tl/t-petIGkR0gc

brada4, 2023-11-20:
Unexpected keyboard by Juloo permits to override "next" key with "enter" and the value gets saved with "enter", but default " next " activity does not update input field.