openwrt/luci #6585: luci-mod-status: wan port flagged as not part of a network but actually in pppoe

Ansuel, 2023-09-22:
## Steps to reproduce:
Make a wan port to provide a pppoe connection and notice that status space show the port as "not part of any network"
Network tab correctly display it as part of the wan network.

[image]

[image]

## Actual behavior:
[image]

## Expected behavior:
On hover the wan port is correctly detected part of the wan network

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
[LuCI Master (git-23.236.53281-98e3743)](https://github.com/openwrt/luci) / [OpenWrt SNAPSHOT (r23995-ce7209bd21)](https://openwrt.org/)
```

McGiverGim, 2023-09-28:
Strange, in my case is a PPPoE connection too, and it seems to be detected as part of the WAN network:

[image]

Ansuel, 2023-09-28:
@McGiverGim can you post your network config maybe we can compare?

McGiverGim, 2023-09-28:
Sure, I think this is what you are asking for:
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd94:XXXXX::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ipaddr '192.168.100.1'
        option ip6assign '60'
        option ip6ifaceid '::1'
        option ip6hint '00'

config interface 'wan'
        option device 'wan.20'
        option proto 'pppoe'
        option username 'XXXXXXXX'
        option password 'XXXXXXXX'
        option peerdns '0'
        option ipv6 '0'
        list dns '127.0.0.1'

config interface 'wan6'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'
        option peerdns '0'
        option device '@wan'
        list dns '::1'

config device
        option type '8021q'
        option ifname 'wan'
        option vid '20'
        option name 'wan.20'

config interface 'guest'
        option proto 'static'
        option netmask '255.255.255.0'
        option device 'br-inv'
        option ipaddr '192.168.101.1'
        option ip6assign '60'
        option ip6hint '10'

config device 'guest_dev'
        option type 'bridge'
        option name 'br-inv'

```

Ansuel, 2023-09-28:
And this is mine...

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'
	option ula_prefix 'fdb3:365c:5cd6::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'

config device
	option name 'lan1'

config device
	option name 'lan2'

config device
	option name 'lan3'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.2.1'
	option netmask '255.255.255.0'
	option ip6weight '1000'
	option delegate '0'
	option ip6assign '64'

config device
	option name 'wan'

config interface 'wan'
	option device 'wan'
	option proto 'pppoe'
	option username 'ewsdfuigqoi'
	option password 'ewoihewigw'
	option ipv6 'auto'
	option type 'bridge'
```

jow-, 2023-10-31:
@Ansuel - try removing the `option type bridge` from your wan interface section.

Ansuel, 2023-10-31:
@jow- just tested and can confirm it does fix the problem. But is it correct?

jow-, 2023-11-01:
Yeah, due to multiple reasons:
1) `option type bridge` in the `interface` sections is not really supported anymore, bridges should be declared as devices
2) Your "wan bridge" neither declares "list ports wan" nor "option ifname wan" (the legacy notation for bridge ports) so LuCI can't make the connection from logical network wan -> device br-wan -> dsa port wan

Ansuel, 2023-11-01:
@jow- it's an old config and wasn't aware of these change.

Can't confirm it does work with the following configuration

config device
	option name 'wan'
	option macaddr '34:98:b5:0f:ab:d0'

```
config interface 'wan'
	option device 'wan'
	option proto 'pppoe'
	option username 'ewsdfuigqoi'
	option password 'ewoihewigw'
	option ipv6 'auto'
	option type 'bridge'
	list ports 'wan'
```

jow-, 2023-11-01:
Hm, I think we don't implement support for old style bridges at all in LuCI anymore. I'll close it anyway since it works as intended with a corrected config.

closed 2023-11-01 by jow-