openwrt/luci #6595: "Unsupported package(s): button-hotplug, kernel"

Asurin219, 2023-09-25:
# Problem
Maybe something wrong with [firmware-selector](https://firmware-selector.openwrt.org/).When I tried to build custom firmware on firmware-selector,I get the error below,switching versions also doesn't work(I didn't have this problem yesterday):
``Unsupported package(s): button-hotplug, kernel``

and the following are the software packages entered during the custom build:
```
luci luci-base luci-compat luci-i18n-base-zh-cn
luci-i18n-firewall-zh-cn
luci-i18n-opkg-zh-cn
vsftpd openssh-sftp-server
htop curl
ttyd luci-app-ttyd luci-i18n-ttyd-zh-cn
```
This is the screenshot with the error:
[image]

jow-, 2023-09-25:
Not a LuCI issue.

closed 2023-09-25 by jow-

hnyman, 2023-09-25:
Not a LuCI issue, but could this be fallout from 
https://github.com/openwrt/openwrt/commit/658549837282d0ca9decfdaf599fc67bf90418c6
as the failing packages gpio-button-hotplug, kernel, mac80211 are built from kernel sources, and do not have a proper package files. (well, mac80211 has Makefile, but is done from kernel backports)

cc @feckert @hauke @aparcar 

Forum discussion:
* https://forum.openwrt.org/t/firmware-selector-wdr4300-v1-non-existing-default-packages-listed/172564

aparcar, 2023-09-25:
Yea this broke the ImageBuilder but I'm working on it. Thanks for pinging me.