openwrt/luci #6605: luci-app-acme: does not show the certificate list

McGiverGim, 2023-09-29:
## Steps to reproduce:
1. Go to the ACME app in luci.

## Actual behavior:
The list of certificates is Empty.

## Expected behavior:
The list of certificates must contain the certicates availables.

## Additional Information:
Debugging the javascript, the problem seems to be here:
https://github.com/openwrt/luci/blob/f51f52bfbb4fba77cc72548bc36908b2697c8c0a/applications/luci-app-acme/htdocs/luci-static/resources/view/acme.js#L9

It seems the `fs.list('/etc/ssl/acme/')` is not returning the files of the folder. I think luci moved from server to javascript, maybe this is the culprit? There is some replacement for this method to execute the command in the router?

vehoelite, 2023-09-30:
Thanks man. I thought this was a misconfiguration on my end. Probably saved me a few days trying to pinpoint why the ACME script was behaving this way. This would be a extremely convenient way of obtaining server/router certificates if I can get it working.... is there a work around / fix for this? any other methods as easy and straightforward as this anywhere?
Side note: The crooks @ godaddy want $70year, no less than 3 years and require $210, no less to get started.!!

McGiverGim, 2023-10-01:
I can't say. @stokito I think you made the code. Some idea?

stokito, 2023-10-01:
In the new version of acmesh package was made [many changes to simplify configuration](https://github.com/openwrt/packages/pull/17721):
1. All certificates are issued into the /etc/ssl/acme/ folder. You don't need to configure it.
2. After issue a hotplug hook is called. Webservers should setup their hook and reload themselves.
3. For webroot challenge always used /var/run/acme folder. Webservers should always server the folder on the path /.well-known/acme/challenges/

For the Luci app I added the list of generated certificates from the /etc/ssl/acme/ folder. If you updated the acme-luci-app and don't see any certs there... then there are n't any. If you have some old acme configurations with different location then the generated certs are there. You just won't see them here.

Please provide more details. Does https works on your sever? Or it was broken.
Which type of a challenge you are using.

McGiverGim, 2023-10-01:
I've certificates in the default /etc/ssl/acme folder. I don't have configured the https server.
I'm on latest master. Do you test it on latest master? I suspect that the move to JavaScript broke it, but maybe I'm wrong.

stokito, 2023-10-01:
@McGiverGim can you please call me in telegram stokito or matrix strokito and show your screen?

McGiverGim, 2023-10-01:
I'm sorry, I don't have telegram or similar. I post here some screen captures if it helps:

1. The certificates are not shown:
[image]

2. This is the content of the folder:
[image]

3. This is the javascript debug, as you see entries is empty:
[image]

stokito, 2023-10-01:
Ok, that's great that the certs are really in place and this is only UI issue.
Could you please open Inspect/ Network tab and check the request to ubus with the files list call?
That should be a JSON-RPC call with POST method and the response must contain an error code.

Are you using the default uhttpd webserver or maybe Lighttpd?
The uhttpd has mod-ubus which provides /ubus endpoint but for the Lighttpd the ubus module is absent. So some api calls may be processd by a Luci legacy ubus endpoint but some may use the direct ubus call.

Another possible reason is that somethibg wrong with the ACL rules.
Did you installed the luci-app-acme with opkg or copied manually?

McGiverGim, 2023-10-01:
Now I'm out of home. I will do the requested tests tomorrow.
I've the default webserver, and installed acme from the UI, software tab.

stokito, 2023-10-01:
Watch for the POST request to http://192.168.1.1/cgi-bin/luci/admin/ubus
Its response should look like:
```json
[
    {
        "id": 2,
        "jsonrpc": "2.0",
        "result": [
            0,
            {
                "entries": [
                    {
                        "type": "file",
                        "inode": 70006,
                        "ctime": 1685838206,
                        "atime": 1685838206,
                        "uid": 0,
                        "mtime": 1685838206,
                        "gid": 0,
                        "mode": 33188,
                        "name": "*.stokito.duckdns.org.crt",
                        "size": 1655
                    },
                    {
                        "type": "file",
                        "inode": 70003,
                        "ctime": 1685838201,
                        "atime": 1685838201,
                        "uid": 0,
                        "mtime": 1685838201,
                        "gid": 0,
                        "mode": 33188,
                        "name": "*.stokito.duckdns.org.key",
                        "size": 288
                    },
                    {
                        "type": "file",
                        "inode": 70007,
                        "ctime": 1685838206,
                        "atime": 1685838206,
                        "uid": 0,
                        "mtime": 1685838206,
                        "gid": 0,
                        "mode": 33188,
                        "name": "*.stokito.duckdns.org.fullchain.crt",
                        "size": 5406
                    },
                    {
                        "type": "file",
                        "inode": 70009,
                        "ctime": 1685838206,
                        "atime": 1685838206,
                        "uid": 0,
                        "mtime": 1685838206,
                        "gid": 0,
                        "mode": 33152,
                        "name": "*.stokito.duckdns.org.combined.crt",
                        "size": 5694
                    },
                    {
                        "type": "file",
                        "inode": 70008,
                        "ctime": 1685838206,
                        "atime": 1685838206,
                        "uid": 0,
                        "mtime": 1685838206,
                        "gid": 0,
                        "mode": 33188,
                        "name": "*.stokito.duckdns.org.chain.crt",
                        "size": 3751
                    }
                ]
            }
        ]
    }
]
```

McGiverGim, 2023-10-02:
It seems it makes two petitions at same request. No error, no list of certificates:

Petition:
```json
[
    {
        "jsonrpc": "2.0",
        "id": 1,
        "method": "call",
        "params": [
            "121e7fe364a0d45c6879c21e6610699d",
            "uci",
            "changes",
            {}
        ]
    },
    {
        "jsonrpc": "2.0",
        "id": 2,
        "method": "call",
        "params": [
            "121e7fe364a0d45c6879c21e6610699d",
            "file",
            "list",
            {
                "path": "/etc/ssl/acme/"
            }
        ]
    }
]
```
Response:
```json
[
    {
        "jsonrpc": "2.0",
        "id": 1,
        "result": [
            0,
            {
                "changes": {}
            }
        ]
    },
    {
        "jsonrpc": "2.0",
        "id": 2,
        "result": [
            0,
            {
                "entries": []
            }
        ]
    }
]
```

Can I execute the same petition using `ubus` directly on the router?

stokito, 2023-10-02:
On the router you can execute:

ubus call file list '{path": "/etc/ssl/acme/"}'

Then it should return the list of files.

It's ok that two json-rpc request were merged into on http request. That's optimization.

McGiverGim, 2023-10-02:
Strange, it does not show any certificate:
```
root@router-comedor:~# ubus call file list '{"path": "/etc/ssl/acme/"}'
{
        "entries": [

        ]
}
```

but the `ls` command does. Maybe some permission problem in the folder with the certificates?

McGiverGim, 2023-10-02:
```
ls -ltr /etc/ssl/

drwxr-xr-x    1 root     root           624 Sep 24 13:04 acme
```
It seems the certificates inside are "links" and not "files", maybe this is the problem?
```
lrwxrwxrwx    1 root     root            62 Sep 24 13:04 xxxx.mywire.org.key -> /root/.acme//xxxx.mywire.org/xxxx.mywire.org.key
lrwxrwxrwx    1 root     root            49 Sep 24 13:04 xxxx.mywire.org.fullchain.crt -> /root/.acme//xxxx.mywire.org/fullchain.cer
lrwxrwxrwx    1 root     root            62 Sep 24 13:04 xxxx.mywire.org.crt -> /root/.acme//xxxx.mywire.org/xxxx.mywire.org.cer
lrwxrwxrwx    1 root     root            48 Sep 24 13:04 xxxx.mywire.org.combined.crt -> /root/.acme//cccc.mywire.org/combined.cer
lrwxrwxrwx    1 root     root            42 Sep 24 13:04 xxxx.mywire.org.chain.crt -> /root/.acme//xxxx.mywire.org/ca.cer
```

McGiverGim, 2023-10-02:
I think we are getting to something. The files linked in the folder DON'T EXIST at all. I don't know why it is pointing to this router, but are wrong. I suppose something changed some time ago and I didn't notice it.

McGiverGim, 2023-10-02:
I think I understand what happens. `acme` downloads the certificates into folders with the name of the domain in `/etc/acme`. You list only `/etc/acme/ssl`, that are the certificates for the HTTPS server. I don't use the HTTPS server, but I had here some old links that point to missing certs from one old test that I did.

So the actual code will not show for me any certificate.

Thanks for your help and your time!!! At least now I understand the problem.