openwrt/luci #6628: luci-nginx-ssl: Cannot satisfy the following dependencies

mmartinv, 2023-10-09:
Installation of `luci-nginx-ssl` fails in Opewrt 23.05.0-rc4 with the following error:
```
root@bpi-r3:~# opkg install luci-ssl-nginx
Installing luci-ssl-nginx (git-22.299.78093-3260371) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0-rc4/packages/aarch64_cortex-a53/luci/luci-ssl-nginx_git-22.299.78093-3260371_all.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency uwsgi for nginx-mod-luci
 * pkg_hash_check_unresolved: cannot find dependency uwsgi-luci-support for nginx-mod-luci
 * pkg_hash_fetch_best_installation_candidate: Packages for nginx-mod-luci found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl-nginx:
 * 	uwsgi
 * 	uwsgi-luci-support
 * opkg_install_cmd: Cannot install package luci-ssl-nginx.
```

jow-, 2023-10-09:
Not a LuCI issue, seems that uwsgi fails to build upstream.

closed 2023-10-09 by jow-

Ansuel, 2023-10-09:
@jow- correct. Just to give some context, it's related by a rust change that modified how python3 based modules compile and is getting handled. So it's just something to wait.

https://github.com/openwrt/packages/pull/22339 this is the PR that should fix the thing.