openwrt/luci #6636: luci-app-sqm: Latency target, auto vs default

timur-davletshin, 2023-10-17:
Luci-app-sqm -> Dangerous Configuration

Comment for Latency target (ingress, egress) says that "default" has a special meaning but in fact "auto" is used. Clarify comment and make choice between "auto" and "default".

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_ARCH='x86_64'
DISTRIB_TAINTS=''
```

jow-, 2023-10-19:
@tohojo

tohojo, 2023-10-20:
@timur-davletshin I'm not really sure what change you are proposing here? The help text already says "leave empty for automatic selection, put in the word default for the qdisc's default", which is what sqm-scripts does...

timur-davletshin, 2023-10-20:
> @timur-davletshin I'm not really sure what change you are proposing here? The help text already says "leave empty for automatic selection, put in the word default for the qdisc's default", which is what sqm-scripts does...

Yep, but my installation says "auto" in respective text field. I believe it should say "default" since I didn't specify anything else.

hnyman, 2023-10-20:
> but my installation says "auto" in respective text field

Yep, the default config has "auto":

https://github.com/tohojo/sqm-scripts/blob/405aeffa48e6385df4bb267e2a2fac6e254f96f8/platform/openwrt/sqm-uci#L13-L14

timur-davletshin, 2023-10-20:
> > but my installation says "auto" in respective text field
> 
> Yep, the default config has "auto":
> 
> https://github.com/tohojo/sqm-scripts/blob/405aeffa48e6385df4bb267e2a2fac6e254f96f8/platform/openwrt/sqm-uci#L13-L14

So, it should be either "auto" or "default" in both places. If I write "default" will be the same behavior as default "auto"? If it is the same, why two names?

tohojo, 2023-10-23:
timur-davletshin ***@***.***> writes:

> So, it should be either "auto" or "default" in both places. If I write
> "default" will be the same behavior as default "auto"? If it is the
> same, why two names?

'auto' (or just leaving the field empty), means that sqm-scripts will
calculate an appropriate 'target' value based on the bandwidth setting.
'default' means no explicit target will be calculated by sqm-scripts,
and the qdisc will use its default. For sch_cake, the default happens to
be a calculation similar to that done by sqm-scripts, but this is
qdisc-dependent...

timur-davletshin, 2023-10-23:
Description should be clarified in this case. It makes wrong impression that those two are the same.

tohojo, 2023-10-23:
On 23 October 2023 17:09:21 CEST, timur-davletshin ***@***.***> wrote:
>Description should be clarified in this case. It makes wrong impression that those two are the same.

Not the way I read it. But please, feel free to suggest a better wording? :)