openwrt/luci #6648: luci-base: Too much recursion

shaman79, 2023-10-25:
## Steps to reproduce:
1. Upgrade to 23.05
2. Login->Overview not working showing errort "Too much recursion"

## Actual behavior:
Overview not loading
Error in console
Internal error: Too much recursion

[image]

## Expected behavior:
Overview visible

## Additional Information:
Seems that some function is no longer available in 23.05
Console error:
```
RPCError: RPC call to luci/getBuiltinEthernetPorts failed with error -32002: Access denied
  at handleCallReply (https://192.168.1.2/luci-static/resources/rpc.js?v=git-23.292.78378-27fb6e5:11:3)
    raise https://192.168.1.2/luci-static/resources/luci.js?v=git-23.292.78378-27fb6e5:155
    handleCallReply https://192.168.1.2/luci-static/resources/rpc.js?v=git-23.292.78378-27fb6e5:11
[luci.js:161:9](https://192.168.1.2/luci-static/resources/luci.js?v=git-23.292.78378-27fb6e5)
InternalError: too much recursion
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:37
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
[luci.js:161:9](https://192.168.1.2/luci-static/resources/luci.js?v=git-23.292.78378-27fb6e5)
Uncaught (in promise) InternalError: too much recursion
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:37
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
    resolveVLANPorts https://192.168.1.2/luci-static/resources/view/status/include/29_ports.js?v=git-23.292.78378-27fb6e5:39
[luci.js:37:12](https://192.168.1.2/luci-static/resources/luci.js?v=git-23.292.78378-27fb6e5%20line%20178%20%3E%20eval)
```
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_REVISION='r23497-6637af95aa'
```

shaman79, 2023-10-25:
Here it is...

```
root@CudyX6-2:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option ipv6 '0'
        option stp '1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option netmask '255.255.255.0'
        list dns '192.168.1.1'
        option delegate '0'
        option ipv6 'off'
        option gateway '192.168.1.1'
        option ipaddr '192.168.1.2'

config device
        option name 'eth0'
        option ipv6 '0'

config device
        option name 'lan1'

config device
        option name 'lan2'
        option ipv6 '0'

config device
        option name 'lan3'
        option ipv6 '0'

config device
        option name 'lan4'
        option ipv6 '0'

config device
        option name 'wan'
        option ipv6 '0'

config device 'guest_dev'
        option type 'bridge'
        option name 'br-guest'
        list ports 'br-guest.3'
        option stp '1'

config interface 'guest'
        option proto 'static'
        option netmask '255.255.255.0'
        option type 'bridge'
        option delegate '0'
        option device 'br-guest'
        option ipaddr '192.168.2.2'

```

jow-, 2023-10-25:
The LuCI recursion issue should be fixed/prevented but your config is semantically wrong:

> config device 'guest_dev'
>        option type 'bridge'
>        option name 'br-guest'
>        list ports 'br-guest.3'
>        option stp '1'

You declare a bridge `br-guest` which includes itself as port, this is what is causing the recursion.

shaman79, 2023-10-25:
It was the problem indeed, it is working after configuring the VLAN properly.

commit 25dd8934 (2023-10-25): luci-mod-status: protect against infinite recursion in port status

When attempting to resolve VLAN devices, protect against potential deep
recursion due to invalid bridge configs referencing themselves.

Fixes: #6648
commit f8753674 (2023-11-02): luci-mod-status: protect against infinite recursion in port status

When attempting to resolve VLAN devices, protect against potential deep
recursion due to invalid bridge configs referencing themselves.

Fixes: #6648

(cherry picked from commit 25dd8934f18a6cebfca1adb8a0d4b5d0bed73145)
closed 2023-10-25 by jow-