openwrt/luci #6666: luci-i18n-firewall-de: problem with NAT entrys

ThomasCr, 2023-10-31:
***
NAT entry's are destroyed / displayed in the right way:

[image]

When I switch back to English, the entry works correct.

## Steps to reproduce:
install luci-i18n-firewall-de (if not set up automatically, switch language) and go to `Netzwerk -> Firewall -> NAT-Regeln`

create an rule or use this one:

```
config nat
        option name 'Tmp'
        list proto 'all'
        option src 'mgmt'
        option dest_ip '192.168.0.0/24'
        option target 'MASQUERADE'
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_REVISION='r23497-6637af95aa'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0 r23497-6637af95aa'
DISTRIB_TAINTS=''
```

```
root@gw:~# opkg list-installed | grep '\-de' | grep -e base -e firewall
luci-i18n-base-de - git-23.301.56412-787ab17
luci-i18n-firewall-de - git-23.301.56412-787ab17
```

closed 2023-10-31 by jow-

castillofrancodamian, 2023-11-01:
Not just in that section. Also under "Port Forwarding".

[image]

jow-, 2023-11-01:
Well, that's another language and another string. I suppose someone broke the spanish translation, root cause is probably the same, mismatched opening or closing curly brace.

castillofrancodamian, 2023-11-01:
It's such a long chain that it's confusing 😟

jow-, 2023-11-01:
Yes, adding indentation helps. It should look like this:
```
%{
	ipv6? 
	%{
		ipv4?
			<var>IPv4</var> y <var>IPv6</var>
		:
			<var>IPv6</var>
	}
	:
	<var>IPv4</var>
}
%{
	proto?
		, protocolo 
		%{
			proto#
			%{
				next?
					, 
			}
			%{
				item.types?
					<var class=\"cbi-tooltip-container\">
					%{item.name}
					<span class=\"cbi-tooltip\">ICMP con tipos 
					%{
						item.types#%{next?, }<var>%{item}</var>
					}</span></var>
				:
					<var>
					%{item.name}
					</var>
			}
		}
}
%{ 
	mark?
		, marca <var
		%{
			marca.inv? 
				data-tooltip="Hacer coincidir fwmarks excepto 
				%{mark.num}
				%{
					mark.mask? 
						con máscara 
						%{mark.mask}
				}
				."
			:
				%{
					mark.mask? 
						data-tooltip="Enmascarar el valor de fwmark con 
						%{mark.mask} 
						antes de comparar."
				}
		}
		>
		%{mark.val}
		</var>
}
%{
	helper?
		, helper 
		%{
			helper.inv?
				<var data-tooltip="Coincide con cualquier asistente excepto &quot;
				%{helper.name}
				&quot;"
				>
				%{helper.val}
				</var>
		}
}
```

But the current spanish string has multiple problems. Will fix tomorrow.

castillofrancodamian, 2023-11-01:
I think I've already fixed it in Weblate by copying the original string and translating the words.

commit da202240 (2023-10-31): luci-app-firewall: fix german SNAT traqnslation entry

Fixes: #6666
commit 5eeaaf44 (2023-10-31): luci-app-firewall: fix german SNAT traqnslation entry

Fixes: #6666

(cherry picked from commit da202240bf48e6953fb1ec9f715da1408b734a5c)