openwrt/luci #6673: NETGEAR R8000 Nighthawk there isn`t  WiFi interface and Web Interface Luci

dotignore, 2023-11-04:
Precondition:

```
CFE for Foxconn Router R8000 version: v1.0.8
Build Date: Mon May 22 10:05:24 CST 2017
Init Arena
Init Devs.
Boot up from NAND flash...
Bootcode Boot partition size = 524288(0x80000)
DDR Clock: 800 MHz
Info: DDR frequency set from clkfreq=1000,*800*
et2: Broadcom **BCM47XX** 10/100/1000 Mbps Ethernet Controller 7.14.43.2 (r474543)
CPU type 0x0: 1000MHz
Tot mem: 262144 KBytes
```

BUG 1 critical
openwrt-23.05.0-rc2-bcm53xx-generic-netgear_r8000-squashfs 
there isn`t  WiFi interface in Luci
it similar behavior on firmware
openwrt-22.03.0-bcm53xx-generic-netgear_r8000-squashfs
openwrt-19.07.9-bcm53xx-netgear-r8000-squashfs

BUG 2  critical
openwrt-23.05.0-rc3-bcm53xx-generic-netgear_r8000-squashfs
there isn`t  WiFi interface and Web Interface Luci + no ping 192.168.1.1 lan1, lan2, lan3, lan4
it similar behavior on firmware
openwrt-23.05.0-bcm53xx-generic-netgear_r8000-squashfs

its ok? BCM47XX BCM5301X
et2: Broadcom **BCM47XX** 10/100/1000 Mbps Ethernet Controller 7.14.43.2 (r474543)
Hardware        : **BCM5301X**

```
BusyBox v1.36.1 (2023-06-26 11:20:39 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc2, r23228-cd17d8df2a
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 0 (v7l)
BogoMIPS        : 1000.00
Features        : half thumb fastmult edsp tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc09
CPU revision    : 0

processor       : 1
model name      : ARMv7 Processor rev 0 (v7l)
BogoMIPS        : 1000.00
Features        : half thumb fastmult edsp tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc09
CPU revision    : 0

[... 563 lines cut ...]
[   23.323411] device lan2 entered promiscuous mode
[   23.414748] b53-srab-switch 18007000.ethernet-switch lan3: configuring for phy/gmii link mode
[   23.429635] br-lan: port 3(lan3) entered blocking state
[   23.435024] br-lan: port 3(lan3) entered disabled state
[   23.440733] device lan3 entered promiscuous mode
[   23.533776] b53-srab-switch 18007000.ethernet-switch lan4: configuring for phy/gmii link mode
[   23.548158] b53-srab-switch 18007000.ethernet-switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   23.558107] br-lan: port 4(lan4) entered blocking state
[   23.563466] br-lan: port 4(lan4) entered disabled state
[   23.569963] device lan4 entered promiscuous mode
[   23.662737] br-lan: port 4(lan4) entered blocking state
[   23.668015] br-lan: port 4(lan4) entered forwarding state
[   23.683482] b53-srab-switch 18007000.ethernet-switch wan: configuring for phy/gmii link mode
[   23.700734] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   23.878335] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/usb2
[   23.886001] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/usb2
[   23.893687] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/usb2
[   23.901278] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/usb2
[   23.924267] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/usb3
[   23.931899] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/usb3
[   23.939450] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/usb3
[   23.947085] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/usb3
```

[image]

dotignore, 2023-11-04:
openwrt-23.05.0-rc3-bcm53xx-generic-netgear_r8000-squashfs

```
Digital core power voltage set to 1.0V
Decompressing...done
Digital core power voltage set to 1.0V

SHMOO VER 1.13

PKID07DC06011801080000000000001A103F01000000

S300001BB
000011B0

RDLYW0 00000004

RDENW0 00000030

RDQSW0

    0000000000111111111122222222223333333333444444444455555555556666
    0123456789012345678901234567890123456789012345678901234567890123
 00 ----+++++++++++++++++++++++++X++++++++++++++++++++++++----------
 01 -------++++++++++++++++++++++++X++++++++++++++++++++++++--------
 02 ----++++++++++++++++++++++++X++++++++++++++++++++++++-----------
 03 ----+++++++++++++++++++++++X+++++++++++++++++++++++-------------
 04 -+++++++++++++++++++++++X+++++++++++++++++++++++----------------
 05 -----+++++++++++++++++++++++X+++++++++++++++++++++++------------
 06 --+++++++++++++++++++++++X++++++++++++++++++++++----------------
 07 -------++++++++++++++++++++++++X++++++++++++++++++++++++--------
 08 ---+++++++++++++++++++++++X++++++++++++++++++++++---------------
 09 ---------+++++++++++++++++++++++X+++++++++++++++++++++++--------
 10 -----+++++++++++++++++++++++X+++++++++++++++++++++++------------
 11 ------------+++++++++++++++++++++++X+++++++++++++++++++++++-----
 12 -----+++++++++++++++++++++++X++++++++++++++++++++++-------------
 13 -----------++++++++++++++++++++++X++++++++++++++++++++++--------
 14 ------+++++++++++++++++++++++X+++++++++++++++++++++++-----------
 15 -----------++++++++++++++++++++++++X+++++++++++++++++++++++-----

PW0

    0000000000111111111122222222223333333333444444444455555555556666
    0123456789012345678901234567890123456789012345678901234567890123
 00 ----+++++++++++++++++++++++++X+++++++++++++++++++++++++---------
 01 -------+++++++++++++++++++++++++X+++++++++++++++++++++++++------
 02 ----++++++++++++++++++++++++X++++++++++++++++++++++++-----------
 03 ---+++++++++++++++++++++++++X+++++++++++++++++++++++++----------
 04 ++++++++++++++++++++++++X++++++++++++++++++++++++---------------
 05 ----+++++++++++++++++++++++++X++++++++++++++++++++++++----------
 06 --++++++++++++++++++++++++X+++++++++++++++++++++++--------------
 07 ------+++++++++++++++++++++++++X+++++++++++++++++++++++++-------
 08 ++++++++++++++++++++++++X++++++++++++++++++++++++---------------
 09 ------+++++++++++++++++++++++++X++++++++++++++++++++++++--------
 10 ----++++++++++++++++++++++++X+++++++++++++++++++++++------------
 11 ----------++++++++++++++++++++++++X++++++++++++++++++++++++-----
 12 --+++++++++++++++++++++++++X++++++++++++++++++++++++------------
 13 --------++++++++++++++++++++++++X+++++++++++++++++++++++--------
 14 ---+++++++++++++++++++++++++X++++++++++++++++++++++++-----------
 15 --------+++++++++++++++++++++++++X+++++++++++++++++++++++++-----

NW0

    0000000000111111111122222222223333333333444444444455555555556666
    0123456789012345678901234567890123456789012345678901234567890123
 00 -----++++++++++++++++++++++++X++++++++++++++++++++++++----------
 01 -------++++++++++++++++++++++++X+++++++++++++++++++++++---------
 02 -----++++++++++++++++++++++++X+++++++++++++++++++++++-----------
 03 ----+++++++++++++++++++++++X+++++++++++++++++++++++-------------
 04 -+++++++++++++++++++++++X+++++++++++++++++++++++----------------
 05 -----+++++++++++++++++++++++X+++++++++++++++++++++++------------
[... 615 lines cut ...]
[   25.182922] device lan2 entered promiscuous mode
[   25.274157] b53-srab-switch 18007000.ethernet-switch lan3: configuring for phy/gmii link mode
[   25.287067] br-lan: port 3(lan3) entered blocking state
[   25.292423] br-lan: port 3(lan3) entered disabled state
[   25.298040] device lan3 entered promiscuous mode
[   25.404810] b53-srab-switch 18007000.ethernet-switch lan4: configuring for phy/gmii link mode
[   25.422332] b53-srab-switch 18007000.ethernet-switch lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   25.431923] br-lan: port 4(lan4) entered blocking state
[   25.437222] br-lan: port 4(lan4) entered disabled state
[   25.443600] device lan4 entered promiscuous mode
[   25.533180] br-lan: port 4(lan4) entered blocking state
[   25.538441] br-lan: port 4(lan4) entered forwarding state
[   25.558594] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   25.580818] b53-srab-switch 18007000.ethernet-switch wan: configuring for phy/gmii link mode
[   25.635274] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   25.643283] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   25.651214] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   25.659113] leds bcm53xx:white:usb2: Failed to get trigger sources for /leds/led-usb2
[   25.682627] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   25.690627] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   25.698571] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
[   25.706555] leds bcm53xx:white:usb3: Failed to get trigger sources for /leds/led-usb3
```

[image]

dotignore, 2023-11-04:
Firmware Version 1.0.4.84 from https://www.downloads.netgear.com/files/GDC/R8000/R8000-V1.0.4.84_10.1.84.zip

https://www.netgear.com/support/product/r8000#download

```
Digital core power voltage set to 1.0V
Decompressing...done
Digital core power voltage set to 1.0V

SHMOO VER 1.13

PKID07DC06011801080000000000001A103F01000000

S300001B7
000011A0

RDLYW0 00000004

RDENW0 00000030

RDQSW0

    0000000000111111111122222222223333333333444444444455555555556666
    0123456789012345678901234567890123456789012345678901234567890123
 00 -----++++++++++++++++++++++++X++++++++++++++++++++++++----------
 01 -------++++++++++++++++++++++++X+++++++++++++++++++++++---------
 02 -----++++++++++++++++++++++++X+++++++++++++++++++++++-----------
 03 ----++++++++++++++++++++++++X+++++++++++++++++++++++------------
 04 -+++++++++++++++++++++++X+++++++++++++++++++++++----------------
 05 -----+++++++++++++++++++++++X+++++++++++++++++++++++------------
 06 ---++++++++++++++++++++++X++++++++++++++++++++++----------------
 07 -------++++++++++++++++++++++++X++++++++++++++++++++++++--------
 08 ---+++++++++++++++++++++++X++++++++++++++++++++++---------------
 09 ---------+++++++++++++++++++++++X+++++++++++++++++++++++--------
 10 -----+++++++++++++++++++++++X+++++++++++++++++++++++------------
 11 -----------++++++++++++++++++++++++X+++++++++++++++++++++++-----
 12 -----+++++++++++++++++++++++X++++++++++++++++++++++-------------
 13 ----------+++++++++++++++++++++++X++++++++++++++++++++++--------
 14 -----++++++++++++++++++++++++X+++++++++++++++++++++++-----------
 15 ----------++++++++++++++++++++++++X++++++++++++++++++++++++-----

PW0

    0000000000111111111122222222223333333333444444444455555555556666
    0123456789012345678901234567890123456789012345678901234567890123
 00 ----+++++++++++++++++++++++++X+++++++++++++++++++++++++---------
 01 -------+++++++++++++++++++++++++X+++++++++++++++++++++++++------
 02 ----++++++++++++++++++++++++X++++++++++++++++++++++++-----------
 03 ----+++++++++++++++++++++++++X++++++++++++++++++++++++----------
 04 ++++++++++++++++++++++++X++++++++++++++++++++++++---------------
 05 -----++++++++++++++++++++++++X++++++++++++++++++++++++----------
 06 -++++++++++++++++++++++++X+++++++++++++++++++++++---------------
 07 ------+++++++++++++++++++++++++X+++++++++++++++++++++++++-------
 08 ++++++++++++++++++++++++X++++++++++++++++++++++++---------------
 09 ------+++++++++++++++++++++++++X++++++++++++++++++++++++--------
 10 ---++++++++++++++++++++++++X++++++++++++++++++++++++------------
 11 ----------++++++++++++++++++++++++X++++++++++++++++++++++++-----
 12 --++++++++++++++++++++++++X++++++++++++++++++++++++-------------
 13 --------++++++++++++++++++++++++X+++++++++++++++++++++++--------
 14 ---+++++++++++++++++++++++++X++++++++++++++++++++++++-----------
 15 --------+++++++++++++++++++++++++X+++++++++++++++++++++++++-----

NW0

    0000000000111111111122222222223333333333444444444455555555556666
    0123456789012345678901234567890123456789012345678901234567890123
 00 -----++++++++++++++++++++++++X+++++++++++++++++++++++-----------
 01 -------++++++++++++++++++++++++X+++++++++++++++++++++++---------
 02 -----++++++++++++++++++++++++X+++++++++++++++++++++++-----------
 03 ----+++++++++++++++++++++++X+++++++++++++++++++++++-------------
[... 1571 lines cut ...]
url_retrieve: Executing 'curl -s -m 180 -o /tmp/circleinfo.txt --cacert /opt/xagent/certs/ca-bundle-mega.crt https://http.fw.updates1.netgear.com/sw-apps/parental-control/circle/r8000/https//circleinfo.txt'
Daemon is down or you are running an older version of it
failed to init bd library
Daemon is down or you are running an older version of it
failed to init bd library
Re try in 5 sec, waiting for the agent to load properly
trying to read xagent.pid ...
Could not open xagent.pid
Daemon is down or you are running an older version of it
failed to init bd library
Daemon is down or you are running an older version of it
failed to init bd library
Daemon is down or you are running an older version of it
failed to init bd library
url_retrieve: Executing 'curl -s -m 180 -o /tmp/circleinfo.txt --cacert /opt/xagent/certs/ca-bundle-mega.crt https://http.fw.updates1.netgear.com/sw-apps/parental-control/circle/r8000/https//circleinfo.txt'
Daemon is down or you are running an older version of it
failed to init bd library
Daemon is down or you are running an older version of it
failed to init bd library
Re try in 5 sec, waiting for the agent to load properly
trying to read xagent.pid ...
Could not open xagent.pid
Daemon is down or you are running an older version of it
failed to init bd library
```

dotignore, 2023-11-04:
open console 

u: admin
p: password

```
bin    dev    lib    mnt    proc   share  tmp    var
data   etc    media  opt    sbin   sys    usr    www

1                      18164                  bus
10                     18165                  cmdline
11                     18170                  conflict_hijack
146                    18232                  cpu
17425                  18252                  cpuinfo
17498                  18271                  crypto
17500                  18289                  devices
17502                  18302                  diskstats
17505                  18303                  dmu
17609                  18838                  driver
17612                  194                    emf
17619                  19452                  execdomains
17627                  19577                  filesystems
17645                  19578                  fs
17647                  2                      interrupts
17654                  218                    iomem
17659                  290                    ioports
17672                  295                    ipv6_wan_DAD_detected
17677                  3                      irq
17731                  300                    kallsyms
17732                  305                    kmsg
17775                  310                    kpagecount
17817                  315                    kpageflags
17827                  320                    loadavg
17829                  325                    locks
17831                  330                    meminfo
17833                  335                    misc
17839                  338                    modules
17845                  341                    mounts
17847                  346                    mtd
17850                  351                    nat_stats
17854                  356                    net
17855                  359                    net_status
17861                  362                    pagetypeinfo
17863                  367                    partitions
17914                  372                    printer_status
17926                  4                      scsi
17941                  413                    self
17942                  418                    slabinfo
17943                  5                      softirqs
17944                  6                      stat
17970                  7                      swaps
17976                  72                     sys
18084                  74                     sysrq-trigger
18086                  75                     sysvipc
18090                  84                     timer_list
18091                  9                      tty
18147                  9109                   uptime
18149                  9112                   version
18159                  9115                   vmallocinfo
18160                  NetUSB                 vmstat
18161                  bcm947xx               wan_status
18162                  brcmnand               zoneinfo
18163                  buddyinfo

/proc # cat cpuinfo
Processor       : ARMv7 Processor rev 0 (v7l)
processor       : 0
BogoMIPS        : 1998.84

processor       : 1
[... 213 lines cut ...]
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
6: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP300> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9c:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
7: dpsta: <BROADCAST,MULTICAST> mtu 1500 qdisc noop
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9c:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
9: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9c:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
10: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9c:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
11: vlan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP300> mtu 1500 qdisc noqueue
    link/ether 9c:3d:cf:xx:x:xx brd ff:ff:ff:ff:ff:ff
12: vlan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue
    link/ether 9c:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
13: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue
    link/ether 9c:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
14: wl0.1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9e:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
15: wl1.1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9e:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
16: wl2.1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 9e:3d:cf:xx:xx:xx brd ff:ff:ff:ff:ff:ff
/proc # Daemon is down or you are running an older version of it
```

hnyman, 2023-11-05:
Closing as this is not related to the LuCI GUI.

The issue is a duplicate of 
* https://github.com/openwrt/openwrt/issues/13784#issuecomment-1779998860
* https://github.com/openwrt/openwrt/issues/13548

and is documented on the 23.05.0 release notes:

* https://openwrt.org/releases/23.05/notes-23.05.0#known_issues

>Known issues
> ...
> bcm53xx: Netgear R8000 and Linksys EA9200 Ethernet is broken (see: https://github.com/openwrt/openwrt/issues/13548)

closed 2023-11-05 by hnyman