openwrt/luci #6691: BUG:The Chinese interface servers of 23.05 do not display Chinese.

Tonkercke, 2023-11-13:
After the system was updated to openwrt23.05.0 and the system language was switched to Chinese, the servers option did not change and was still in English.

I tried a full reset of the openwrt system settings, but the problem persists.

[image]

systemcrash, 2023-11-13:
You're welcome to contribute a translation if one does not yet exist today. 
https://hosted.weblate.org/projects/openwrt/luci/zh_Hant/
https://hosted.weblate.org/projects/openwrt/luci/zh_Hans/

yurtpage, 2024-01-09:
The Sevices menu item is translated for many years: 
https://github.com/openwrt/luci/blob/master/modules/luci-base/po/zh_Hans/base.po#L8294
https://github.com/openwrt/luci/blob/master/modules/luci-base/po/zh_Hant/base.po#L8303

@Tonkercke Please confirm that you still have the problem

jow-, 2024-02-18:
This should work fine. What is odd about the OP is the lowercase spelling of the "services" menu, the original english menu title is capitalized. Maybe some unofficial 3rd party packaged added this untranslated menu entry.

closed 2024-02-18 by jow-