openwrt/luci #6694: Commit access

systemcrash, 2023-11-13:
There's still a bunch I don't yet know, but I'd be happy to help with some of the load in this repo. 

Perhaps openwrt/openwrt at a later date although I'm still mainly just lurking/opining in there. 

Ping @jow- @hnyman @Ansuel @aparcar @feckert

stangri, 2023-11-13:
PRs by @systemcrash for convenience: https://github.com/openwrt/luci/pulls?q=is%3Apr+author%3Asystemcrash

feckert, 2023-11-15:
I would say :+1: . But that's for @jow- to decide, as he has the admin rights.

aparcar, 2023-11-20:
@openwrt/luci-admin ping

aparcar, 2023-12-03:
I invited @systemcrash to group with commit access to luci.git 

@systemcrash thanks for your contribution and please enjoy OpenWrt

closed 2023-12-03 by aparcar

Ansuel, 2023-12-03:
Afaik CODEOENERS is only for enterprise plan of GitHub

systemcrash, 2023-12-03:
@aparcar as I have been for many years :)

@Ansuel : second paragraph (I didn't know this either):

You can define code owners in public repositories with GitHub Free and GitHub Free for organizations, and in public and private repositories with GitHub Pro, GitHub Team, GitHub Enterprise Cloud, and GitHub Enterprise Server. For more information, see "[GitHub’s plans](https://docs.github.com/en/get-started/learning-about-github/githubs-plans)."

Ansuel, 2023-12-03:
Oh they changed that? I remember noticing the feature but was only for paid plan... Nice! Would be good to use on packages

systemcrash, 2023-12-03:
A code owner seems to have implications. They know the code best, but when they're busy or absent, and they're the only one who can approve, some PRs tend to hang there for a while. ☹️ 

Maybe openwrt will break this rule: CODEOWNERS files must be under 3 MB in size 😄

stokito, 2024-01-13:
> A CODEOWNERS seems to have implications. When they're busy, and they're the only one who can approve, some PRs tend to hang there for a while.

The CODEOWNERS will just request for a reviewer automatically. It's safe and similar to current approach of mentioning a maintainer manually to ask for a review.

The problem is that the review should be a member of the repository but also have a `Write` permission. Otherwise it can be used for a spam attack: you can mention anyone and they'll be annoyed by PRs.

The `Write` permission may allow to commit directly to a `master` branch. That's why the branch protection should be enabled were only maintainers can change it.

So please merge the PR #6830