openwrt/luci #6696: No bridge in "RADIUS VLAN Tagged Interface"

j-d-r, 2023-11-13:
Hello,

I'm trying to setup RADIUS automatic VLAN but the selection list does not allows bridges for "RADIUS VLAN Tagged Interface". See [HERE IN CODE](https://github.com/openwrt/luci/blob/55e0f28b1ec2f86068d1dad9915641e3fb22afbf/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js#L1459C24-L1459C24).

My setup is a bridge (br-vlan) in device tab with VLAN in "bridge VLAN filtering" tab. It creates automatically br-vlan.XXX devices.

In "RADIUS VLAN Tagged Interface" field I must choose br-vlan but it's not available because it's a bridge. Only br-vlan.XXX are available but choosing will add device br-vlan.XXX.YYY to bridge.

Why bridges are forbidden ? Disabling the JS check make my setup work.

systemcrash, 2023-11-13:
Adding bridges might work. This was implemented before DSA, so RADIUS resulting VLANS were thought to be `eth0.23`. But if it works, perhaps we can allow bridges here. 

If you add e.g. `br-vlan` to your config, does RADIUS add a per-user VLAN as intended to the `br-vlan` bridge?

j-d-r, 2023-11-13:
Yes, it works fine, two clients can successfully connect at the same time in two VLAN.

`brctl show` looks something like:
```
bridge name       bridge id               STP enabled     interfaces
br-vlan-rad23     8000.cafecafecafe       no              br-vlan.23
                                                          wl1-ap1.23

br-vlan-rad24     8000.cafecafecafe       no              br-vlan.24
                                                          wl1-ap1.24
```

By the way, it's off-topic, I used br-vlan bridge because I didn't found a way to just use `eth0.23` or `wan.23`. I can create interfaces with `wan.23` that obtains IP in VLAN 23 but inside a hostapd created bridge DHCP is not working for wifi clients. I can see `wan.23` in bridge but clients never receive DHCP replies. It might be a DSA problem/limitation, or more probably a configuration error.

commit 26e4b69a (2023-11-19): luci-mod-network: wireless.js: allow bridges for radius vlan tagged ifaces

Fixes: #6696
commit 27793bd1 (2023-11-19): luci-mod-network: wireless.js: allow bridges for radius vlan tagged ifaces

Fixes: #6696

(cherry picked from commit 26e4b69abf0ce97197dea07b9e7c7f944b4c9e7c)
closed 2023-11-19 by jow-