openwrt/luci #6702: luci-app-vnstat2: XHR request timed out accessing config

jkugler, 2023-11-17:
## Steps to reproduce:
I'm using vnstat2 to track bandwidth usage, and it's working great.

Going to /cgi-bin/luci/admin/status/vnstat2/graphs shows all my graphs, and everything looks good.

However, if I go to '/cgi-bin/luci/admin/status/vnstat2/config, I get an error: "XHR request timed out" and it will not display the config.

I've looked through the logs, and haven't seen any messages from vnstat2 or Luci. 

I can upload the DB file somewhere if you wish.

## Actual behavior:
XHR timeout

## Expected behavior:
Displays configuration screen

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS='
```
vnstat2 config:
```
# cat /etc/config/vnstat

config vnstat
        list interface '6in4-wan6'
        list interface 'pppoe-wan'
```
and the conf file:
```
# cat /etc/vnstat.conf
# vnStat 2.9 configuration file
#
# lines starting with # or ; are comments, everything has default
# values, remove ; before each option to change its value

# default interface (leave empty for automatic selection)
;Interface ""

# location of the database directory
DatabaseDir "/mnt/sda1/vnstat"

# locale (LC_ALL) ("-" = use system locale)
;Locale "-"

# date output formats for -d, -m, -t and -w
;DayFormat    "%Y-%m-%d"
;MonthFormat  "%Y-%m"
;TopFormat    "%Y-%m-%d"

# characters used for visuals
;RXCharacter       "%"
;TXCharacter       ":"
;RXHourCharacter   "r"
;TXHourCharacter   "t"

# how units are prefixed when traffic is shown
# 0 = IEC standard prefixes (KiB/MiB/GiB...)
# 1 = old style binary prefixes (KB/MB/GB...)
# 2 = SI decimal prefixes (kB/MB/GB...)
;UnitMode 0

# used rate unit (0 = bytes, 1 = bits)
;RateUnit 1

# how units are prefixed when traffic rate is shown in bits
# 0 = IEC binary prefixes (Kibit/s...)
# 1 = SI decimal prefixes (kbit/s...)
;RateUnitMode 1

# output style
# 0 = minimal & narrow, 1 = bar column visible
# 2 = same as 1 except rate in summary
# 3 = rate column visible
;OutputStyle 3
;EstimateBarVisible 1

# number of decimals to use in outputs
;DefaultDecimals 2
;HourlyDecimals 1

# spacer for separating hourly sections (0 = none, 1 = '|', 2 = '][', 3 = '[ ]')
;HourlySectionStyle 2

# how many seconds should sampling for -tr take by default
;Sampletime 5

# default query mode
# 0 = normal, 1 = days, 2 = months, 3 = top, 5 = short
# 7 = hours, 8 = xml, 9 = one line, 10 = json
;QueryMode 0

# default list output entry limits (0 = all)
;List5Mins      24
;ListHours      24
;ListDays       30
;ListMonths     12
;ListYears       0
;ListTop        10

# how to match interface given for query to interface in database
# 0 = case sensitive exact match to interface name
# 1 = method 0 followed by case sensitive exact match of alias
# 2 = method 1 followed by case insensitive exact match of alias
# 3 = method 2 followed by case insensitive beginning match of alias
;InterfaceMatchMethod 3

# vnstatd
##

# switch to given user when started as root (leave empty to disable)
;DaemonUser ""

# switch to given group when started as root (leave empty to disable)
;DaemonGroup ""

# try to detect interface maximum bandwidth, 0 = disable feature
# MaxBandwidth will be used as fallback value when enabled
;BandwidthDetection 1

# maximum bandwidth (Mbit) for all interfaces, 0 = disable feature
# (unless interface specific limit is given)
;MaxBandwidth 1000

# interface specific limits
#  example 8Mbit limit for eth0 (remove # to activate):
#MaxBWeth0 8

# data retention durations (-1 = unlimited, 0 = feature disabled)
;5MinuteHours   48
;HourlyDays      4
;DailyDays      62
;MonthlyMonths  25
;YearlyYears    -1
;TopDayEntries  20

5MinuteHours   -1
HourlyDays     -1
DailyDays      -1                                                
MonthlyMonths  -1            
YearlyYears    -1          
TopDayEntries  -1

# how often (in seconds) interface data is updated
;UpdateInterval 20

# how often (in seconds) interface status changes are checked
;PollInterval 5

# how often (in minutes) data is saved to database
;SaveInterval 5

# how often (in minutes) data is saved when all interface are offline
;OfflineSaveInterval 30

# rescan database after save for new interfaces to be monitored (1 = enabled, 0 = disabled)
;RescanDatabaseOnSave 1

# automatically start monitoring all interfaces not found in the database
# (1 = enabled, 0 = disabled)
;AlwaysAddNewInterfaces 0

# on which day should months change
;MonthRotate 1
;MonthRotateAffectsYears 0

# filesystem disk space check (1 = enabled, 0 = disabled)
;CheckDiskSpace 1

# how much the boot time can variate between updates (seconds)
;BootVariation 15

# create database entries even when there is no traffic (1 = enabled, 0 = disabled)
;TrafficlessEntries 1

# how many minutes to wait during daemon startup for system clock to
# sync time if most recent database update appears to be in the future
;TimeSyncWait 5

# how often (in minutes) bandwidth detection is done when
# BandwidthDetection is enabled (0 = disabled)
;BandwidthDetectionInterval 5

# force data save when interface status changes (1 = enabled, 0 = disabled)
;SaveOnStatusChange 1

# enable / disable logging (0 = disabled, 1 = logfile, 2 = syslog)
;UseLogging 2

# create dirs if needed (1 = enabled, 0 = disabled)
;CreateDirs 1

# update ownership of files if needed (1 = enabled, 0 = disabled)
;UpdateFileOwner 1

# file used for logging if UseLogging is set to 1
;LogFile "/var/log/vnstat/vnstat.log"

# file used as daemon pid / lock file
;PidFile "/var/run/vnstat/vnstat.pid"

# 1 = 64-bit, 0 = 32-bit, -1 = old style logic, -2 = automatic detection
;64bitInterfaceCounters -2

# use SQLite Write-Ahead Logging mode (1 = enabled, 0 = disabled)
;DatabaseWriteAheadLogging 0

# change the setting of the SQLite "synchronous" flag
# (-1 = auto, 0 = off, 1, = normal, 2 = full, 3 = extra)
;DatabaseSynchronous -1

# database uses UTC instead of local timezone (1 = enabled, 0 = disabled)
;UseUTC 0

# vnstati
##

# title timestamp format
;HeaderFormat "%Y-%m-%d %H:%M"

# show hours with rate (1 = enabled, 0 = disabled)
;HourlyRate 1

# show rate in summary (1 = enabled, 0 = disabled)
;SummaryRate 1

# transparent background (1 = enabled, 0 = disabled)
;TransparentBg 0

# image size control
;LargeFonts 0
;LineSpacingAdjustment 0
;ImageScale 100

# 5 minutes graph size control
;5MinuteGraphResultCount 576
;5MinuteGraphHeight 300

# hourly graph mode (0 = 24 hour sliding window, 1 = begins from midnight)
;HourlyGraphMode 0

# horizontal/vertical summary graph (0 = hours, 1 = 5 minutes)
;SummaryGraph 0

# traffic estimate bar style
# (0 = not shown, 1 = continuation of existing bar, 2 = separate bar)
;EstimateStyle 1

# bar column in list outputs shows rate if OutputStyle is 3
# (1 = enabled, 0 = disabled)
;BarColumnShowsRate 0

# image colors
;CBackground     "FFFFFF"
;CEdge           "AEAEAE"
;CHeader         "606060"
;CHeaderTitle    "FFFFFF"
;CHeaderDate     "FFFFFF"
;CText           "000000"
;CLine           "B0B0B0"
;CLineL          "-"
;CRx             "92CF00"
;CTx             "606060"
;CRxD            "-"
;CTxD            "-"
```

janh, 2023-11-17:
I can't reproduce this issue, not even using OpenWrt 22.03.5 on a ipq40xx device with your config.

> However, if I go to '/cgi-bin/luci/admin/status/vnstat2/config, I get an error: "XHR request timed out" and it will not display the config.

Have you tried using your browser's developer tools to find the request that is timing out?

> I can upload the DB file somewhere if you wish.

Does the database actually make a difference? (changing the DatabaseDir option temporarily is probably the easiest way to test)

jkugler, 2023-11-17:
I don't know...I just thought maybe it was trying to process something in the DB and was failing to do it in time.

I don't get any errors in developer tools, just this in the browser:
[image]

This is what the Network tab looks like after that error pops up:
[image]

I see vnstat2 chew about 25% CPU for a while during graph generation, then when I go to config, it chews 25% CPU for a while, and continues to do so for a little bit after the error pops up.

`vnstat --json f 1` from the prompt takes about 23 seconds. How long is the XHR timeout? How might I go about upping that timeout?

jkugler, 2023-11-17:
According to `/www/luci-static/resources/view/vnstat2/config.js` the command `fs.exec('/usr/bin/vnstat',['--json','f','1'])` is run at some point. Maybe load? (not a JS coder, sorry):
```
o.load=function(section_id){return fs.exec('/usr/bin/vnstat',['--json','f','1'])
```

janh, 2023-11-17:
> `vnstat --json f 1` from the prompt takes about 23 seconds.

This command is only used to get the list of interfaces in the database. However, it actually also loads the last five-minute entry, as well as the total for every interface. I assume the total is why it is so slow for you, as your configuration is to keep all data forever (so there are probably a lot of five-minute entries in the database).

Can you try if the commands `vnstat --json y 1` (which fetches yearly data) or `vnstat --dbiflist 1` are faster?

Then I could change it to use one of these commands instead (the `--dbiflist` option did not exist yet when the code was originally written).

> How long is the XHR timeout?

My guess from the last request in your second screenshot, is that the timeout is 20 seconds.

jkugler, 2023-11-17:
```
# time vnstat --json y 1
{"vnstatversion":"2.11","jsonversion":"2","interfaces":[{"name":"6in4-wan6","alias":"","created":{"date":{"year":2023,"month":10,"day":26},"timestamp":1698345800},"updated":{"date":{"year":2023,"month":11,"day":17},"time":{"hour":14,"minute":50},"timestamp":1700265000},"traffic":{"total":{"rx":777401321545,"tx":45876122112},"year":[{"id":3,"date":{"year":2023},"timestamp":1672563600,"rx":777401321545,"tx":45876122112}]}},{"name":"6rd-wan6","alias":"","created":{"date":{"year":2023,"month":4,"day":7},"timestamp":1680898185},"updated":{"date":{"year":2023,"month":10,"day":26},"time":{"hour":9,"minute":11},"timestamp":1698340300},"traffic":{"total":{"rx":2409166198808,"tx":417873708349},"year":[{"id":2,"date":{"year":2023},"timestamp":1672563600,"rx":2409166198808,"tx":417873708349}]}},{"name":"pppoe-wan","alias":"","created":{"date":{"year":2023,"month":4,"day":7},"timestamp":1680898185},"updated":{"date":{"year":2023,"month":11,"day":17},"time":{"hour":14,"minute":50},"timestamp":1700265000},"traffic":{"total":{"rx":6712229635292,"tx":1136924080960},"year":[{"id":1,"date":{"year":2023},"timestamp":1672563600,"rx":6712229635292,"tx":1136924080960}]}}]}
real    0m 0.03s
user    0m 0.01s
sys     0m 0.01s
```
Just a little faster. :)
```
# time vnstat --dbiflist 1
6in4-wan6
6rd-wan6
pppoe-wan
real    0m 0.02s
user    0m 0.01s
sys     0m 0.00s
```
Even faster. :)

janh, 2023-11-17:
Thank you, I'll create a pull request later.

jkugler, 2023-11-18:
Excellent, thank you!

PR #6704 (2023-11-19): luci-app-vnstat2: use more efficient way to get database interfaces
The currently used command "vnstat --json f 1" can be very slow when the database contains many entries. This can result in an XHR timeout error on the configuration page.

Use the "--dbiflist" option instead, which was added in vnStat 2.7.

Fixes: #6702

---

This should also be cherry-picked for 23.05 and 22.03.
commit 3ac4f567 (2023-11-19): luci-app-vnstat2: use more efficient way to get database interfaces

The currently used command "vnstat --json f 1" can be very slow when the
database contains many entries. This can result in an XHR timeout error
on the configuration page.

Use the "--dbiflist" option instead, which was added in vnStat 2.7.

Fixes: #6702
commit 2762cd8e (2023-11-19): luci-app-vnstat2: use more efficient way to get database interfaces

The currently used command "vnstat --json f 1" can be very slow when the
database contains many entries. This can result in an XHR timeout error
on the configuration page.

Use the "--dbiflist" option instead, which was added in vnStat 2.7.

Fixes: #6702

(cherry picked from commit 3ac4f567dfdfd6ee989ceffb025241e605790b6d)
commit 74dd23bd (2023-11-19): luci-app-vnstat2: use more efficient way to get database interfaces

The currently used command "vnstat --json f 1" can be very slow when the
database contains many entries. This can result in an XHR timeout error
on the configuration page.

Use the "--dbiflist" option instead, which was added in vnStat 2.7.

Fixes: #6702

(cherry picked from commit 3ac4f567dfdfd6ee989ceffb025241e605790b6d)
closed 2023-11-19 by jow-