openwrt/luci #6721: Question About Configuration ( With Pictures )

lxnunorodrigues, 2023-11-26:
Hi 

Hi have : Raspberry Pi Compute Module 4 Rev 1.0  ( 192.168.0.1 )
Firmware : OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-23.306.39416-c86c256

DNS Pihole 
1 - 192.168.0.171
2 - 192.168.0.170

My Network IP : 192.168.0.1 

I have servers and cameras on range 192.168.0.2 / .120

Today i installed in option "sofware" OpenConnect VPN 

I Use on Android phone : AnyConnect "Cisco"

Now i can see my network and navigate on internet , but i have 2 Pihole´s DNS  , how configure if conected to VPN and use that to remove ADS ?

I dont understand what i configure , firewall etc... 

if possible atach "images" for easy

i use zerotier on another pi with pihole ,no problems .

thanks

hnyman, 2023-11-26:
Please discuss on the support forum.

closed 2023-11-26 by hnyman

lxnunorodrigues, 2023-11-26:
hi , where is ?

lxnunorodrigues, 2023-11-26:
i think ok , i see , i public same text ,thanks

hnyman, 2023-11-26:
https://forum.openwrt.org/