openwrt/luci #6773: What is the purpose of this line of code?

Cdreamfly, 2023-12-20:
## What is the purpose of this line of code?
[modules/luci-base/htdocs/luci-static/resources/rpc.js](https://github.com/openwrt/luci/blob/f6ce0b4362b5c268527631228b5703d5660d42c6/modules/luci-base/htdocs/luci-static/resources/rpc.js#L38)

```
return request.post(rpcBaseURL + q, req, {
    timeout: (L.env.rpctimeout || 20) * 1000,  //What is the purpose of this line of code?  The timeout is always 1000?
    nobatch: nobatch,
    credentials: true
}).then(cb, cb);
```

jow-, 2023-12-20:
Ift sets the timeout to the configured amount of seconds (or 20) times 1000 to yield a millisecond value.

closed 2023-12-20 by jow-

Cdreamfly, 2023-12-20:
> Ift sets the timeout to the configured amount of seconds (or 20) times 1000 to yield a millisecond value.

Is this JavaScript syntax? I don't really understand.

Cdreamfly, 2023-12-20:
> > Ift sets the timeout to the configured amount of seconds (or 20) times 1000 to yield a millisecond value.
> 
> Is this JavaScript syntax? I don't really understand.

Is || a logical operator?

Cdreamfly, 2023-12-20:
> Ift sets the timeout to the configured amount of seconds (or 20) times 1000 to yield a millisecond value.

> Ift sets the timeout to the configured amount of seconds (or 20) times 1000 to yield a millisecond value.

Thank you for your answer. I have no doubts.