openwrt/luci #6781: A large amount of untranslated content appears in the dhcp list

try496, 2023-12-23:
Built based on the latest 23.05, a large number of translation failures occurred. I also tried other languages ​​and had the same problem. This problem occurred after the recent synchronization tree. All the languages ​​I tried have been translated.

[image]
[image]

try496, 2023-12-23:
[image]
[image]

try496, 2023-12-23:
After testing, it was found that some of the previous synchronizations were gone after this synchronization. Some of them existed in the translation file but were not applied.

try496, 2023-12-23:
After comparing the synchronized files, I found that some directories in the translation files no longer exist after this synchronization. Some are in the translation files but have not been applied.

try496, 2023-12-23:
[image]
[image]
[image]

try496, 2023-12-23:
[image]

try496, 2023-12-23:
[image]

try496, 2023-12-23:
[image]

hnyman, 2023-12-23:
Well, like I answered to you in discussion at https://github.com/openwrt/luci/pull/6775#issuecomment-1867651826 , all translations are made by volunteers like you and me.

New strings are not automatically translated, but require manual work. So, it may take a while after changes until you (or somebody else) does the translation and the new strings translations get merged (first into main/master). And then later they get backported into stable 23.05 every few weeks.

There has recently been lots of small string tweaking and string inclusion style changes by @systemcrash , so it is quite possible that there is more than normal amount of untranslated strings.

I just backported all current translations to 23.05, so hopefully at least some of the missing strings now appear also there.
But not all. 
E.g. that string `"Dump cache on SIGUSR1, include requesting IP."` that you are showing as example, is still untranslated in main/master:

https://hosted.weblate.org/translate/openwrt/luci/zh_Hans/?checksum=09746f26e807153a
https://hosted.weblate.org/translate/openwrt/luci/zh_Hant/?checksum=09746f26e807153a
[image]

and is then naturally also untranslated in 23.05:

```
+#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js:326
+msgid "Dump cache on SIGUSR1, include requesting IP."
+msgstr ""
```

The strings really need an actual translation by somebody before they can automagically appear on the end-user screen.

try496, 2023-12-23:
Because the later parts have been abandoned in the main branch, I rolled back to the December 16 version and all translations are normal. Based on the latest main branch, all translations are also normal.---Original---From: "Hannu ***@***.***>Date: Sat, Dec 23, 2023 15:31 PMTo: ***@***.***>;Cc: ***@***.***>***@***.***>;Subject: Re: [openwrt/luci] 【23.05】A large amount of untranslated content appears in the dhcp list (Issue #6781)
Well, like I answered to you in discussion at #6775 (comment) , all translations are made by volunteers like you and me.
New strings are not automatically translated, but require manual work. So, it may take a while after changes until you (or somebody else) does the translation and the new strings translations get merged (first into main/master). And then later they get backported into stable 23.05 every few weeks.
There has recently been lots of small string tweaking and string inclusion style changes by @systemcrash , so it is quite possible that there is more than normal amount of untranslated strings.
I just backported all current translations to 23.05, so hopefully at least some of the missing strings now appear also there.
But not all.
E.g. that string "Dump cache on SIGUSR1, include requesting IP." that you are showing as example, is still untranslated in main/master:
https://hosted.weblate.org/translate/openwrt/luci/zh_Hans/?checksum=09746f26e807153a
https://hosted.weblate.org/translate/openwrt/luci/zh_Hant/?checksum=09746f26e807153a
image.png (view on web)
and is then naturally also untranslated in 23.05:
+#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js:326
+msgid "Dump cache on SIGUSR1, include requesting IP."
+msgstr ""

The strings really need an actual translation by somebody before they can automagically appear on the end-user screen.

try496, 2023-12-23:
The reason may be that the developer changed the text about translation in dhcp.js, but it was not changed in 22.05, resulting in translation errors after synchronization.

try496, 2023-12-23:
After simultaneous translation today, I found some letter strings on the translation website, but there are still some letter strings that were not found. Since 23.05 may take quite a long time to be translated, I have rolled back to the officially released version.

try496, 2023-12-23:
> 好吧，就像我在[#6775（评论）](https://github.com/openwrt/luci/pull/6775#issuecomment-1867651826)的讨论中回答你一样，所有的翻译都是由像你我这样的志愿者完成的。
> 
> 新字符串不会自动翻译，但需要手动操作。因此，更改后可能需要一段时间，直到您（或其他人）进行翻译并且新的字符串翻译被合并（首先合并到 main/master 中）。然后他们每隔几周就会被重新移植到稳定的 23.05。
> 
> 最近有很多小的字符串调整和字符串包含样式更改，因此很可能有超过正常数量的未翻译字符串。
> 
> 我刚刚将所有当前翻译向后移植到 23.05，所以希望至少一些缺失的字符串现在也出现在那里。但不是全部。例如，您作为示例显示的字符串在 main/master 中仍未翻译：`"Dump cache on SIGUSR1, include requesting IP."`
> 
> https://hosted.weblate.org/translate/openwrt/luci/zh_Hans/?checksum=09746f26e807153a https://hosted.weblate.org/translate/openwrt/luci/zh_Hant/?checksum=09746f26e807153a ![image](https://private-user-images.githubusercontent.com/7926856/292614278-9ced0ca5-c41c-494d-b1bf-aa6e9b7d4c89.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzMjM2NTAsIm5iZiI6MTcwMzMyMzM1MCwicGF0aCI6Ii83OTI2ODU2LzI5MjYxNDI3OC05Y2VkMGNhNS1jNDFjLTQ5NGQtYjFiZi1hYTZlOWI3ZDRjODkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjNUMDkyMjMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDU0OWNkNGMxMTU5M2I5NjM3MDJkYTA1MmRhYmI0OGMxNTY5ODY0NTFhNzdlMzQ0NjFlN2I5ZmJiZDc5NTFjOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.xk-FrdF-ZCxQ6Y74UPFkQ1lU1Xfo__1ODaUQNeDxGuI)
> 
> 然后自然也在 23.05 中未翻译：
> 
> ```
> +#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js:326
> +msgid "Dump cache on SIGUSR1, include requesting IP."
> +msgstr ""
> ```
> 
> 这些字符串确实需要有人进行实际翻译，然后才能自动出现在最终用户屏幕上。

> 好吧，就像我在[#6775（评论）](https://github.com/openwrt/luci/pull/6775#issuecomment-1867651826)的讨论中回答你一样，所有的翻译都是由像你我这样的志愿者完成的。
> 
> 新字符串不会自动翻译，但需要手动操作。因此，更改后可能需要一段时间，直到您（或其他人）进行翻译并且新的字符串翻译被合并（首先合并到 main/master 中）。然后他们每隔几周就会被重新移植到稳定的 23.05。
> 
> 最近有很多小的字符串调整和字符串包含样式更改，因此很可能有超过正常数量的未翻译字符串。
> 
> 我刚刚将所有当前翻译向后移植到 23.05，所以希望至少一些缺失的字符串现在也出现在那里。但不是全部。例如，您作为示例显示的字符串在 main/master 中仍未翻译：`"Dump cache on SIGUSR1, include requesting IP."`
> 
> https://hosted.weblate.org/translate/openwrt/luci/zh_Hans/?checksum=09746f26e807153a https://hosted.weblate.org/translate/openwrt/luci/zh_Hant/?checksum=09746f26e807153a ![image](https://private-user-images.githubusercontent.com/7926856/292614278-9ced0ca5-c41c-494d-b1bf-aa6e9b7d4c89.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzMjM2NTAsIm5iZiI6MTcwMzMyMzM1MCwicGF0aCI6Ii83OTI2ODU2LzI5MjYxNDI3OC05Y2VkMGNhNS1jNDFjLTQ5NGQtYjFiZi1hYTZlOWI3ZDRjODkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjNUMDkyMjMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDU0OWNkNGMxMTU5M2I5NjM3MDJkYTA1MmRhYmI0OGMxNTY5ODY0NTFhNzdlMzQ0NjFlN2I5ZmJiZDc5NTFjOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.xk-FrdF-ZCxQ6Y74UPFkQ1lU1Xfo__1ODaUQNeDxGuI)
> 
> 然后自然也在 23.05 中未翻译：
> 
> ```
> +#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js:326
> +msgid "Dump cache on SIGUSR1, include requesting IP."
> +msgstr ""
> ```
> 
> 这些字符串确实需要有人进行实际翻译，然后才能自动出现在最终用户屏幕上。

I saw on the forum that you said that the stable version is based on the secondary development of the main branch. I saw that the stable branch has a lot of untranslated content. I thought there was an error. I solved some based on the latest compilation, but there are still about 10 texts that have not been translated. 23.05 may take quite some time to synchronize. I have switched back to the main branch to use. I just checked the translation website and saw that there are a lot of new content updates.

closed 2023-12-23 by try496