openwrt/luci #6786: How to have a night LED mode (timer) for router's led to turn off and on all leds at particular times daily on openwrt firmware?

blue-pine, 2023-12-25:
I tried myself and also searched everywhere for setting night led mode feature for OpenWrt firmware but all effort went to no avail for such a simple feature. This feature is already avilable on the stock firmware of my wifi router TP-Link Archer C6 AC1200 v3.2 which is itself based on OpenWrt and for which I have provide all related files from my research here in my repository : https://github.com/blue-pine/openwrt-tplink_archer_c6_v3.2 . Still I have failed to join the led_night_mode library (from stock router firmware) to the OpenWrt firmware for use on the router. So, I ask for everyone's help whatever they can provide for getting this last and only feature lacking on all OpenWrt firmwares. If you need any help then please ask me, you are always welcome.
Thank you

systemcrash, 2023-12-25:
All components are there, LED config, and cron.

blue-pine, 2023-12-26:
> All components are there, LED config, and cron.

Thank you, but I am a beginner and dont know how to go forward, so how can I integrate this timer with openwrt ?

brada4, 2023-12-31:
Forums have more help.
https://github.com/openwrt/openwrt/issues/14279
As a beginner you can quickly test what someone else derives from wifischedule, like cron scripts. Or switch off all leds. Or use electricians tape to hide blinking.

blue-pine, 2024-01-01:
> Forums have more help. [openwrt/openwrt#14279](https://github.com/openwrt/openwrt/issues/14279) As a beginner you can quickly test what someone else derives from wifischedule, like cron scripts. Or switch off all leds. Or use electricians tape to hide blinking.

Happy new year and thank you for your help. Will explore and check for more help

closed 2024-01-08 by feckert