openwrt/luci #6787: local time is incorrect

hiepvq, 2023-12-25:
i set timezone to Asia/Ho Chi Minh but it always incorrect 1 hour compared to current time of my pc
[image]

## Additional Information:
OpenWrt version information 

[Powered by LuCI Master (git-23.346.45702-3e8eb4a)](https://github.com/openwrt/luci) / [ArgonTheme v1.8.3](https://github.com/jerrykuku/luci-theme-argon) / OpenWrt Li [2023] v7.5 Compiled

systemcrash, 2023-12-25:
Does the timezones file or package need updating, I wonder.

What does your `/etc/TZ` contain?

hiepvq, 2023-12-25:
switch to other timezone then switch to my timezone, it display is ok but if reboot router, it will get back is incorrect. that time is chinese

[image]

hiepvq, 2023-12-25:
vi /etc/config/system
[image]

systemcrash, 2023-12-25:
Not that file....

hiepvq, 2023-12-25:
I think config is still works. it only display incorrect local time in System -> system
it still display correct here. my time UTC+7
[image]

hiepvq, 2023-12-25:
> Not that file....

it is empty, there is no this file

brada4, 2023-12-30:
/etc/TZ should be a symlink to /tmp/TZ, which hsould be something along lines of `+7`
`date` vs `date -u` should confirm it.
You are using a fork "compiled by li" , probably you should choose between asking li and using real OpenWRT, absent TZ file kind of says Li removed it.

hiepvq, 2023-12-30:
> /etc/TZ should be a symlink to /tmp/TZ, which contains something along lines of `+7`
> `date` vs `date -u` to confirm it.
> You are using a fork "compiled by li" , probably you should choose between asking li and using real OpenWRT, absent TZ file kind of says Li removed it.

Ok Thank, so i will close this issue

closed 2023-12-30 by hiepvq