openwrt/luci #6789: [Feature Request] luci-app-attendedsysupgrade: LOG of packages

lars18th, 2023-12-26:
Hi @aparcar ,

This is a request related/similar to #6312 but not the same. Here the PROBLEM:

- When you use the "luci-app-attendedsysupgrade" a LOG the packages requested to be introduced in the firmware image aren't saved. Therefore you can't manually reproduce the firmware.

Another question:
- If you want to get the firmware to upload it to other identical devices, then you can't. Because the downloaded image is not provided (a simple link will be sufficient; or an automatic download).

Therefore I suggest (if you agree), to generate a LOG file and store it in a preserved directory (i.e. "/etc/attendedsysupgrade/log/").
Futhermore, perhaps you can provide inside the interface a box to edit the list of the packages.

I hope you want to consider this.
Regards.

brada4, 2023-12-30:
`auc -n` in command line provides link to cached custom image.

lars18th, 2024-01-01:
> `auc -n` in command line provides link to cached custom image.

Nice! But this is not executed when you use the LUCI app. So, I suggest to add this enhancement to the LUCI interface. And also to preserve the LOG.

Regards.

brada4, 2024-01-01:
no idea what are you talking about.server side log?

systemcrash, 2024-01-01:
Normal sys upgrade can stash a log of installed packages text file.

brada4, 2024-01-01:
@aparcar is it possible to add download links for config backup/package list/image file download before hitting yes on attended sysupgrade?

lars18th, 2024-01-02:
> no idea what are you talking about.server side log?

With the current implementation of the LUCI interface you can only do: click&install. Nothing more. However, using the `auc` CLI tool you can do more. For example, you can obtain the link of the firmware; or you can see the list of packages inside the request, etc. The idea is to store all of this data inside a LOG file that will be preserved over the update. This will improve the update of two or three similar devices that we can have at home.

brada4, 2024-01-02:
id like to get hard copy of firmware from the web for the record before applying.