openwrt/luci #6797: luci: en/disable wifi also applies unrelated, queued changes

knarrff, 2023-12-30:
Today I locked myself out of a box and had to use failsafe to get back in because I queued up changes related to vlans and before I was done (did not press 'apply') with that I disabled one of my wifi stations using the buttons on the wifi configuration page. Those buttons apparently do a 'queue change, apply change without 90 seconds revert', as **all** my changes were applied and no rollback occurred after 90 s, locking me out of the machine.

## Steps to reproduce:
1. Make a change that pushes changes into a queue to be applied with the 'apply' button later.
2. Switch to the wireless configuration page
3. enable/disable any station using the button

## Actual behavior:
1. See all queued config applied, without revert after 90s of non-connection.

## Expected behavior:
1. enable/disable the selected interface, but leave all the other changes queued (not applied)

## Additional Information:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_REVISION='r23497-6637af95aa'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0 r23497-6637af95aa'
DISTRIB_TAINTS=''

systemcrash, 2023-12-30:
I ran into this once before and forgot exactly what it was I did to trigger this. Good you logged this. 

I can confirm that this exists.

jow-, 2024-01-05:
Try this fix:
```diff
diff --git a/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js b/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js
index 80ab426a0e..18c03249cf 100644
--- a/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js
+++ b/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js
@@ -260,7 +260,7 @@ function network_updown(id, map, ev) {
        }
 
        return map.save().then(function() {
-               ui.changes.apply()
+               ui.changes.apply(true)
        });
 }
```

systemcrash, 2024-02-02:
@knarrff Have you been able to verify the proposed fix?

knarrff, 2024-02-02:
Sorry, this went below my radar. I'll try it - it does look promising from the look of it.

systemcrash, 2024-02-15:
Have you tried this fix yet @knarrff

systemcrash, 2024-02-15:
> Try this fix:
>
> ```diff
> diff --git a/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js b/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js [...]

@jow- I could not verify any behaviour difference when repeatedly testing this change on 23.05.2.

I lined up some harmless changes, and pressing enable/disable on the radio of a wifi network applied the changes immediately.

jow-, 2024-02-18:
@systemcrash - this patch is not supposed to change that, it merely enables the same rollback protection as the normal apply button. Since wifi enabling/disabling is implemented as writing the `option disabled` setting into the wireless config and issuing a config reload, there's no way to isolate this operation from other pending uci changes.

systemcrash, 2024-02-22:
> it merely enables the same rollback protection as the normal apply button. 

Not what I experienced. It counted down and applied those changes, no roll-back was available.

jow-, 2024-03-04:
The rollback only happens if connectivity is lost.

knarrff, 2024-09-06:
> Have you tried this fix yet @knarrff

Sorry for the late reply. I tried it but it does not seem the have the desired effect. What I tried:

- edited /overlay/upper/www/luci-static/resources/view/network/wireless.js to add the ```true``` in line 38 (```return map.save().then(function(){ui.changes.apply(true)});}```)
- reboot to make very sure the old version is not cached
- look at the source the browser gets for wireless.js and verify that the new version actually 'made it'.
- connect to the AP via a specific wlan-ap on that AP.
- disable that wlan-network on the wireless page of luci
- get disconnected: wait 90s (and more)
- result: the wireless AP did not come back up
- What I would have expected: after 90s the AP should notice no reconnect, revert the disable and thus, enable the AP again.