openwrt/luci #6799: Interface from ipv6-pd via pppoe cannot be modified

d-navikas, 2023-12-31:
## Steps:

1. go to: Network → Interfaces
2. press Edit on wan_6 interface

## Actual behavior:
TypeError
Cannot read properties of null (reading '.name')

[image]

## Expected behavior:
Opened interface settings page or no unusable buttons.

## Additional Information:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS='

network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd5c:1d0a:9d8f::/48'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.@device[1]=device
network.@device[1].name='wan'
network.@device[1].macaddr='5c:02:14:ed:a1:b0'
network.wan=interface
network.wan.device='wan'
network.wan.proto='pppoe'
network.wan.ipv6='auto'
network.wan.username='****
network.wan.password='*****'

closed 2023-12-31 by systemcrash

Ramon00, 2024-01-01:
@systemcrash maybe as an additional note, can we automatically put the virtual interface into the same firewall zone?

systemcrash, 2024-01-01:
It's my understanding that being an implicit part of an existing interface,
they're already covered.

Ramon00, 2024-01-01:
@systemcrash but then color it in the right way in the LuCi view?