openwrt/luci #6800: Some „Weblate“ components are locked due to incorrect configuration.

dziugas1959, 2024-01-02:
Seeing that I requested „sshtunnel“ and „alist“ a month ago to Lithuanian, not to mention some others, that I have missed, but they are simply blocked to be added for translation, due to how it is configured, more in depth discussion https://github.com/openwrt/luci/issues/3183
Simply put, some components are locked to maintainers, so when a component gets added, like „alist“ by a chinese developer, then only his translation is added, and every single other language is locked out. 
Example:
[image]
The best way to fix this, is to add all components to the existing languages, Russian doesn't have „xinetd“, because it is also locked, making the 100% translated quote incorrect.

It's a very easy problem to solve.

Have a nice day, and happy Christmas along with the previous day's new year!

systemcrash, 2024-01-03:
This is not a config issue, rather, neglect. 

I've worked through the luci repo and everything should have po files for all languages now. 

( from at least commit hash 7b445ef406d43ab64911c2b97481873d4333b6e4 )

Have at it 😄

closed 2024-01-03 by systemcrash

try496, 2024-01-03:
Still didn't solve the problem This problem needs to be solved by the maintenance staff of the website I don't know why the OpenVPN translation was emptied

try496, 2024-01-03:
@systemcrash  Please delete the zh file. The translated documents of the Chinese are zh_Hans and zh_Hant

hnyman, 2024-01-03:
Once I get back home after holidays, I will try to figure out why some components seem locked for casual supers, while some are not. It is probably something in the component-specific config. 

There have been lots of new languages for many modules created by casual users, while only a few seem to fail, and cause that "request".  

So, I disagree with @systemcrash opinion, that "it is neglect, not config issue".

Weblate requires quite a few settings per component, so it is quite possible that some components are missing some settings.

Ps. Adding all languages to all components seems unnecessarily extensive, as only a few languages are really actively translated. Most languages have been started, but only small amounts have been translated. Adding those never-to-be-actually-translated languages to many components will just cause us lots of CPU cycle burning at buildbot.

systemcrash, 2024-01-03:
I've probably single-handedly destroyed weblate "translated" stats  😃 

> @systemcrash Please delete the zh file. The translated documents of the Chinese are zh_Hans and zh_Hant

Yeah, well spotted. The fix is in 68d8c0517a7f4ee095603b1185a31059cc007f0f 

> So, I disagree with @systemcrash opinion, that "it is neglect, not config issue".

In any case, the .po(t) files now exist for the remainder of luci projects, so i18n is now at least possible. 

> Ps. Adding all languages to all components seems unnecessarily extensive, as only a few languages are really actively translated. Most languages have been started, but only small amounts have been [...]

I considered this. Although i18n is a continuous effort. At least some languages now have po files that weren't there before, for those popular languages.

systemcrash, 2024-01-03:
@hnyman Perhaps config can be used to prevent more churn for those 'unpopular' languages. But if you wish to share which those are, I could update the luci i18n build tools and prune those out. 

I don't think that would be a great strategy, however.

try496, 2024-01-03:
applications/luciappexample/po/zh_Hans/example.po,  
There is an error with this suffix @systemcrash

try496, 2024-01-03:
In any case, the maintainers have noticed the problem of the translation website. They only need to solve this problem. In the future, users of newly added components will be able to create translation files and translate them by themselves. This problem seems to be traced back to LuCI/applications/smartdns