openwrt/luci #6801: DHCP and DNS settings: dnsmasq domain filter is too restrictive and doesn't allow wildcards

vtavernier, 2024-01-02:
In the Network > DHCP and DNS > CNAME tab, it is possible to edit entries that will be converted into a dnsmasq `--cname` option. In the UI, this is configured with a `hostname` data type for validation:

https://github.com/openwrt/luci/blob/032f3056789d01930028ab5eeecac59c17c7e8d6/modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js#L814

However, dnsmasq supports wildcards in cname aliases, as evidenced by [the docs](https://dnsmasq.org/docs/dnsmasq-man.html):
> [...]
>  --cname as long as the record name is in the authoritative domain. If the target of the CNAME is unqualified, then it is qualified with the authoritative zone name. CNAME used in this way (only) may be wildcards, as in
> 
> --cname=**\*.example.com**,default.example.com

Note that a wildcard domain may be set using the command line (`uci`), and dnsmasq behaves as expected.

The field validation should be relaxed to allow wildcards as well as valid domain names, to allow this feature of dnsmasq to be used.

systemcrash, 2024-02-02:
Hi @vtavernier! In which situations (only) does the manpage document that a CNAME may contain wildcards?

closed 2024-02-18 by jow-

systemcrash, 2024-02-18:
@jow- I was afraid this might happen which was why I asked the question. There is only one very specific use case when this is possible and even then not recommended, and adding wildcard possibility to this field opens up more problems than it solves. One needs to study that part of the man page which upon a few readings is actually easy to misunderstand. As long as we don't advertise this possibility, that's... Acceptable.

jow-, 2024-02-18:
It is allowed by dnsmasq, it is accepted by uci and there's existing configs using it, so there is no reason LuCI should not accept it.

commit c8cddc35 (2024-02-18): luci-mod-network: dhcp.js: relax CNAME domain validation

Relax the CNAME domain validation to allow wildcard names.

Fixes: #6801
commit 947ac215 (2024-03-07): luci-mod-network: dhcp.js: relax CNAME domain validation

Relax the CNAME domain validation to allow wildcard names.

Fixes: #6801

(cherry picked from commit c8cddc35302380882bab910679f717d65f715029)
commit 0e01542e (2024-03-07): luci-mod-network: dhcp.js: relax CNAME domain validation

Relax the CNAME domain validation to allow wildcard names.

Fixes: #6801

(cherry picked from commit c8cddc35302380882bab910679f717d65f715029)