openwrt/luci #6803: luci-mod-network: confusing MTU settings on PPPoE connection.

timur-davletshin, 2024-01-03:
```
DISTRIB_TARGET='x86/geode'
DISTRIB_ARCH='x86'
DISTRIB_DESCRIPTION=openwrt-23.05 branch (git-23.357.58018-024e7ab)/OpenWrt 23.05.2 (r23630-842932a63d)
```

Generally there two ways to setup MTU. One is in Devices tab (Luci UI) and another one is in Interfaces tab (I'm talking about PPPoE connection).
They correspond to respective parts of configuration file: /etc/config/network:

```
config interface 'wan'
	option proto 'pppoe'
	option device 'eth0'
...
	option mtu '1492'

config device
	option name 'pppoe-wan'
	option mtu '1492'
```

## Actual behavior:
Only MTU in "device" section (or Devices tab) will actually apply to PPPoE connection. The other one in "interface" does nothing. Check can be done by ifconfig or directly via Luci UI.

## Expected behavior:
Remove duplicating (and non-functioning) MTU settings in PPPoE interface settings (Interfaces tab).

**Note**: I'm not talking about physical interface (eth0 in this example), only about pppoe-wan.

jow-, 2024-01-04:
Interface mtu option should translate to pppd mtu/mru cli arguments.

systemcrash, 2024-01-05:
The word 'should' is doing a lot of lifting here.

systemcrash, 2024-01-05:
Shouldn't `proto_pppoe_setup()`call to `ppp_generic_setup()` pass `mtu` also?

https://github.com/openwrt/openwrt/blob/f1de1a090fadc74504b3ee1b5bcb914c9110bc6f/package/network/services/ppp/files/ppp.sh#L219-L242

systemcrash, 2024-04-16:
@jow-: for pppoe "should" is false. The MTU option needs to apply to the 'ethernet' interface responsible for the connection, if there is one. 

The manpages for the `pppoe.so` module list zero (direct) options for MTU. 
https://www.man7.org/linux/man-pages/man8/pppd.8.html

So two solutions here are seem evident: either 
a) delete the GUI option, or 
b) remap it so that it applies to the underlying interface (less trivial).

Closed by 995d3a1801cf3b850e8c7c35937fbdd0526ff53e

systemcrash, 2024-04-16:
See also https://github.com/openwrt/openwrt/pull/15176

timur-davletshin, 2024-05-15:
I see luci-proto-ppp has returned MTU settings back. Can someone clarify what does this MTU settings do?

hnyman, 2024-05-15:
> Can someone clarify what does this MTU settings do?

See 
* https://github.com/openwrt/luci/commit/b57c6ff29eb625913b91e0f6ebb754a5454acb9f
* https://github.com/openwrt/openwrt/pull/15176#issuecomment-2107364533

jow-, 2024-05-15:
It is supposed to control the PPP MRU/MTU values which are independent from the underlying ethernet MTU. Due to a bug in netifd this option currently is inoperable, a fix is being worked on.

timur-davletshin, 2024-05-15:
@jow- so the pppd's part working fine it's just 'Devices' tab which is broken?

jow-, 2024-05-15:
No, netifd the ppp protocol handler is broken, the fix for this will not happen in LuCI.