openwrt/luci #6809: luci-app-[adblock-fast|[pbr]|https-dns-proxy] ubus spam?

wulfy23, 2024-01-06:
Effects: 23.05.2 (tested), master?
Maintainer: @stangri

## Steps to reproduce:
1. Install luci-app-[pbr|adblock-fast|https-dns-proxy]

2. They can be enabled or disabled does not matter

3. Insert ubus command wrapper (or monitor using your preferred method)
```
cp /bin/ubus /bin/ubusorig
cat <<PPP >/bin/ubus
#!/bin/sh
echo "$0 $@ wrapper" > /dev/kmsg
echo "PPIDcmd: $(cat /proc/$PPID/cmdline)" > /dev/kmsg
/bin/ubusorig "$@"
PPP
chmod +x /bin/ubus
```
4. Login to luci (main overview page)

5. See lots and lots of runaway? calls trying to get init status which seems unusual/unwanted from here, considering
a) its the overview page
b) these services could all be disabled
c) the calls come in at a pretty high rate and no other services seem to need/exhibit such behaviour

replace your proper ubus command
```
cp /bin/ubusorig /bin/ubus
```

Apologies if this is expected behaviour... just seems kinda out of place to me...

Spam sample / rate for just one of these services
```
[ 4445.606809] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4445.615095] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4445.642779] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4445.651050] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4445.663703] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4445.672084] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4445.684861] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4445.693237] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4445.706508] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4445.715526] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4450.605326] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4450.613607] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4450.641256] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4450.649503] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4450.662163] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4450.670434] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4450.683306] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4450.691749] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4450.704440] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4450.712877] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4455.604300] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4455.612622] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4455.640531] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4455.648848] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4455.661626] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4455.669953] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4455.682813] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4455.691304] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4455.704137] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4455.712500] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4460.608337] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4460.616662] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4460.644444] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4460.652839] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4460.665514] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[... 121 lines cut ...]
[ 4520.682493] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4520.690775] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4520.703490] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4520.711794] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4525.606485] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4525.614781] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4525.642758] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4525.651051] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4525.663827] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4525.672048] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4525.684779] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4525.693157] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4525.705756] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4525.714158] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4530.605550] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4530.613819] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4530.641481] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4530.649822] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4530.662659] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4530.670967] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4530.683832] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4530.692118] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
[ 4530.704706] /bin/ubus call service list { 'name': 'adblock-fast' } wrapper
[ 4530.713180] PPIDcmd: /bin/sh/usr/libexec/rpcd/luci.adblock-fastcallgetInitStatus wrapper
```

stangri, 2024-01-06:
@wulfy23 If this is happening with the version which includes the service status include in the overview page, I'd guess that the getInitStatus calls are due to automatic refresh of the overview.

Do you have any other services which also publish their information to overview?

I'll check if the over overview include files are different from my code tho to see if anything can be improved.

wulfy23, 2024-01-07:
thanks stangri, mwan has some stuff on overview... maybe ddns if thats not default

was not aware of the frontend use ( adblock-fast ) status on overview, makes more sense now

I think other calls go via another path so these ones seemed to really stand out, but **it worked really nicely when starting the service so makes more sense now**

note: tried to reproduce and there was a period where it was showing info and **not excessively making these calls** until service was stopped/disabled so yeah, it doesn't run ALL the time like it appeared to when I first looked.

i'll close this given the above, **this issue sort of diminishes to an almost non-issue**, and if you took a little look at this area, i'm confident you'll fix/improve what if anything can be improved there

(teency but slower/less polling? like double ish so 10 secs? maybe, harmonize path with other services?)

no idea, will leave it to you

thanks for taking a look and apologies for the noise

closed 2024-01-07 by wulfy23

stangri, 2024-01-07:
There is only one call to getInitStatus on render.