openwrt/luci #6811: luci-app-https-dns-proxy: AhaDNS filters do not persist in config

arichiardi, 2024-01-06:
## Steps to reproduce:

1. go to: Services → HTTPS DNS Proxy → Add
2. Select AhaDNS Blitz Provider
3. Insert Filters "1:1.23" - for instance
4. Save, Save & Apply

## Actual behavior:
Luci does not display the saved filters and `cat /etc/config/https-dns-proxy` does not show the filters under the AhaDNS entry.

## Expected behavior:
To persist (and use) the filters.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```

stangri, 2024-01-06:
I can't reproduce, works as expected for me, including proper config file and luci app displaying the filters as well.

```
xg-135r3 in ~ # cat /etc/config/https-dns-proxy

config main 'config'
	option canary_domains_icloud '1'
	option canary_domains_mozilla '1'
	option dnsmasq_config_update '*'
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'
	option procd_trigger_wan6 '0'
	option update_dnsmasq_config '*'

config https-dns-proxy
	option bootstrap_dns '149.112.121.10,149.112.122.10,2620:10A:80BB::10,2620:10A:80BC::10'
	option resolver_url 'https://private.canadianshield.cira.ca/dns-query'

config https-dns-proxy
	option bootstrap_dns '1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'
	option resolver_url 'https://cloudflare-dns.com/dns-query'

config https-dns-proxy
	option resolver_url 'https://blitz.ahadns.com/1:1.23'
```

[image]

arichiardi, 2024-01-06:
Oh that's odd, are you inserting "1:1.23" in the UI and hit Save?
Maybe we are using differenti versions (I will send you mine in a sec).

arichiardi, 2024-01-06:
Using `luci-app-https-dns-proxy 2023-10-25-5 11.36 KiB`

arichiardi, 2024-01-06:
[image]

arichiardi, 2024-01-06:
[image]

stangri, 2024-01-07:
>  The thing that does not work here is when I do not add it initially, then I Edit and modify it later.

I can reproduce that, I'll have a look at it next time I have the time. For now, delete the instance and create it with the params you want.

stangri, 2024-01-09:
I've figured out what's causing it, may take me a while to come up with the fix.

closed 2024-01-09 by stangri

stangri, 2024-01-09:
Thanks for the report, the updated/fixed IPK can be downloaded from my own repository and I'll merge the fixes into openwrt/luci tomorrow.