openwrt/luci #6813: luci-app-mwan3:May I ask about the method of filling in IPv6 when dealing with dynamic IPv6 prefixes?

bcseputetto, 2024-01-07:
[image]

When the IPv6 prefix is dynamic, the IPv6 of the firewall can only fill in the suffix in the format shown in the figure. However, in luci-app-mwan3, neither the source IP nor the destination IP of the rule supports this format. If mwan3 itself supports the format like the firewall, could it be that luci limits this kind of format?

[image]

feckert, 2024-01-08:
I only use IPv4, so I can't check whether this works. If you change the IPv6 address in the configuration file `/etc/config/mwan3` does the mwan3 work as expected? If so, then the LuCI is the limiting factor and the datatype should be changed.

bcseputetto, 2024-01-09:
I need some time to test whether mwan3 works after directly modifying the configuration file. If feasible, I will report back here again.

bupahaibao, 2024-01-09:
> I only use IPv4, so I can't check whether this works. If you change the IPv6 address in the configuration file `/etc/config/mwan3` does the mwan3 work as expected? If so, then the LuCI is the limiting factor and the datatype should be changed.

This does not take effect ，and may also result in the entire configuration file not taking effect

feckert, 2024-01-15:
This is probably due to the fact that `iptables` or `nftables` does not support this kind of IPv6 address type. If you have a fix, you are welcome to create a pull request. I will therefore close the issue.

closed 2024-01-15 by feckert