openwrt/luci #6819: <luci>: Status -> Overview -> Active DHCP Leases: Table sort using header only lasts until next refresh (not working in 23.05)

gb0101010101, 2024-01-09:
## Steps to reproduce:

This was previously reported in #6640 

1. Go to router home page or Status -> Overview.
2.  Scroll down to table of Active DHCP Leases.
3. Click on any of the table headings to sort table.
4. Wait a few seconds for page to refresh, table sort reverts to default.

## Actual behavior:

Table sort works on click but reverts to default sort on next refresh (after a few seconds).

## Expected behavior:

Table sort should be retained at least until page is fully reloaded.

## Additional Information:

Table sort works correctly, being retained, on the Network -> DHCP & DNS -> Static leases page 
`cgi-bin/luci/admin/network/dhcp` in the `Active DHCP Leases` table.

Happens on multiple Target and Arch 

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_REVISION='r23497-6637af95aa'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0 r23497-6637af95aa'
DISTRIB_TAINTS=''
```

jow-, 2024-01-09:
Duplicate of #6640

jow-, 2024-01-09:
The version you're running predates the fix.