openwrt/luci #6838: luci-base/luci-compat: Value optional=false not respect by ui.js

jjm2473, 2024-01-17:
I defined a Value like that:
```lua
o = s:option(Value, "path", translate("Path").."<b>*</b>")
o.rmempty = false
o.optional = false
o.datatype = "string"
o:value("/mnt/abc")
o:value("/mnt/def")
```
I hope this value cannot be empty, even if it is user custom input.

The Value is render to html in here, `optional` is false in that time:
https://github.com/openwrt/luci/blob/7534a0beb429a9bc2f36524f03dcb2ea9eb74bde/modules/luci-compat/luasrc/view/cbi/value.htm#L12

But ui.js will force overwrite `optional` to true, which makes empty value acceptable:
https://github.com/openwrt/luci/blob/7534a0beb429a9bc2f36524f03dcb2ea9eb74bde/modules/luci-base/htdocs/luci-static/resources/ui.js#L2040

If this is a bug, please fix it.

If not, please tell me how to write a Value that has predefined values and accept user input which can not be empty.

systemcrash, 2024-01-17:
https://openwrt.github.io/luci/jsapi/LuCI.ui.Combobox.html

Luci is slowly getting rid of LUA: all new stuff should be js. 

https://github.com/openwrt/luci/blob/7534a0beb429a9bc2f36524f03dcb2ea9eb74bde/modules/luci-base/htdocs/luci-static/resources/ui.js#L2028-L2029C2

> If not, please tell me how to write a Value that has predefined values and accept user input which can not be empty.

There are examples in active use in the code-base. Typically `o.value('one');` etc.

closed 2024-01-17 by systemcrash

McGiverGim, 2024-03-05:
@jjm2473 I'm having a similar problem in my code. Did you fix or found the origin of the problem?

jjm2473, 2024-03-06:
@McGiverGim I didn't fix it. In theory, just deleting this line will do the trick, but I don't know if there will be other problems (in js):
https://github.com/openwrt/luci/blob/7534a0beb429a9bc2f36524f03dcb2ea9eb74bde/modules/luci-base/htdocs/luci-static/resources/ui.js#L2040

OpenWRT should be asked to check this bug.

McGiverGim, 2024-03-06:
My case was a little different, so I finally ended adding `optional = false` AND `rmempty = false`. Adding both variables, fixes the issue in my case (it was a simple input field, not a combo in my case). The strange is that it needs both to make it mandatory.

jjm2473, 2024-03-07:
@McGiverGim "simple input field" does not have this bug.

You also can use `o.datatype = "string"` (or other more suitable datatype) and `o.rmempty = false` to make it mandatory.