openwrt/luci #6858: „LuCI-app-firewall“: First day of the week being incorrect (regional).

dziugas1959, 2024-01-22:
In short, this is more of a cosmetic issue, since „Openwrt“ uses Sunday as the first day of the week („Wikipedia“ – „in the United States, Canada, China, Japan, as well as in parts of South America, Sunday is the first day of the week.“) 
## Steps to reproduce:
Go to Firewall > Traffic Rules > Time Restrictions > Week days. It's very possible (and likely), that other core „LuCI“ do this.

[image]

## Actual behavior:
Week starts on Sunday, while regional or linguistic should make sure, that it's not the case.

## Expected behavior:
Week starts on Monday

## Additional Information:
„OpenWrt“ version: 23.05.2
„LuCI“ version: git-23.051.66410-a505bb1

**Example:**
Depending on the language that the user is using, change the order of the weeks (a.k.a, it just seems weird, that when week days are numerical, it would go seventh day to first day.).
Or add a region option in settings next to time zone (Like in „Microsoft Windows“).

jow-, 2024-01-22:
This is a wontfix, the amount of additional code infrastructure to implement this is too high.

closed 2024-01-22 by jow-

dziugas1959, 2024-01-23:
While I do somewhat understand the effort being more than gain, a simple – „If language is this, then order like this“, the most basic of solutions, though some English speakers will get left out with this approach.