openwrt/luci #6860: What is the error reported after clicking the add button in the node list?Is there a solution?

SHENMIYANG, 2024-01-23:
/usr/lib/lua/luci/i18n.lua:44: bad argument #1 to 'translate' (string expected, got nil)
stack traceback:
	[C]: in function 'translate'
	/usr/lib/lua/luci/i18n.lua:44: in function 'translate'
	...b/lua/luci/model/cbi/passwall/client/node_config.lua:131: in function 'callback'
	/usr/lib/lua/luci/model/uci.lua:289: in function 'foreach'
	...b/lua/luci/model/cbi/passwall/client/node_config.lua:130: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

jow-, 2024-01-23:
It means that the passwall application (which is not part of this repo) has a bug, it calls the `translate()` function with an improper argument. Please report this issue to the creator of the passwall app (I guess immortalwrt)

closed 2024-01-23 by jow-

SHENMIYANG, 2024-01-23:
I don't know much about it.