openwrt/luci #6861: <tcnetemui> include 'tcnetemui' package in luci repository

taisinoneka, 2024-01-23:
Hi, I have created this tools
https://github.com/taisinoneka/tcnetemui
It provides GUI for setting NetEm conditions. I have been able to create a package and install it manually. I'd like to work further on this app to include it in luci codebase. What are the next steps for me?
Note - it is written in Lua using luci-compat dependancy.

systemcrash, 2024-01-25:
Looks like a useful tool - but it won't be accepted until it is converted to js.

taisinoneka, 2024-01-25:
Thanks @systemcrash, I'll try my best to rewrite it to js. Anything else I should take into consideration?

systemcrash, 2024-01-26:
Look around at how other apps and mods in this repo are done in js. Especially recent ones. 

Don't be afraid to make a draft PR and get early feedback and force-push updates to its dev branch.

jow-, 2024-04-20:
Closing this as there's nothing actionable. Feel free to open another issue or PR if you want to move forward with a JavaScript version.

closed 2024-04-20 by jow-